svenhb/GRBL-Plotter #224: Coordinate systems breaking after setting up fiducials

yagizzha, 2021-09-16:
[image]

Coordinate system works correctly until we use fiducials to compensate angle and scaling , at which point they become diffrent . Zero'ing them doesnt take them to their original zero. How is there a step im missing or maybe there is a fix ?

svenhb, 2021-09-16:
When teaching and applying the points, the GCode will be modified - no step back, except reloading the GCode.
https://github.com/svenhb/GRBL-Plotter/wiki/PCB-drilling

Not sure if "Undo" will be shown, as for other operations, via menu - translation

futurau, 2021-09-16:
On main screen DRO shows offset between g54 and machine coordinate system 
but on control coordinate system window we can’t see the offset.

svenhb, 2021-09-16:
Camera form applies G92:
[image]

futurau, 2021-09-16:
https://media.giphy.com/media/3ohhwkfYcpOyeuepqw/giphy.gif

futurau, 2021-09-16:
`https://media.giphy.com/media/3ohhwkfYcpOyeuepqw/giphy.gif`

futurau, 2021-09-16:
Thanks a lot.

closed 2021-09-21 by yagizzha