svenhb/GRBL-Plotter #225: After doing camera setup for tool coordinate, coordinate system breaks

yagizzha, 2021-09-21:
[image]

After 0'ing X and Y to set up our tool and camera coordinate , homing again results in weird points . -711 and -370 on G54 for example . Is there a way to make tool coordinate so that its relative to our current position rather then having a set position?

svenhb, 2021-09-21:
Don't you use homing to get a machine reference point?

svenhb, 2021-09-21:
At least you could move to lower left position (hardware stop) and do an hardware reset of you grbl-controller to get 0;0 for machine coordinates.

futurau, 2021-09-21:
In reality, camera position is relative to tool position (xy offset between camera and tool). 
But camera coordinate system and tool coordinate systems are relative to machine coordinate system in g-code logic.
If we reset grbl or zero machine coordinate system; selected coordinate system (for example G54) change but the other coordinate system doesn’t change (for example G59). 
And the offset value between G54 and G59 became wrong.
We can’t find a way to store and reuse tool and camera offset after homing or resetting system.
Any suggestions?

svenhb, 2021-09-21:
I'm working on two new camera mountings:
- fix camera - observing the whole working area
- camera mounted on xy platform, but pointing on tool position - no need for G54 / G59
Still in progress but I already started the manual: https://grbl-plotter.de/index.php?id=form-camera-1&lang=en

closed 2021-09-30 by yagizzha