svenhb/GRBL-Plotter #226: Potential addition for automation

yagizzha, 2021-09-21:
We've been using your code's changed version together with python to calculate how to automate the process of cutting/drawing . To do this we've used circle fiducials on a checkboard table . Python side changes angle of camera to near 90 degree from top vision and finds fiducials on paper after image has been processed. After taking these and knowing the total size of table it can tell where to go in GCode which we send through a server we put on your program . 

If any of this would be helpful to you i can post them  

Base:
[image]

Top to bottom : 
[image]

Fiducials marked:
[image]

GCode output for top circle first then the bottom one (Top=Empty inside , Bottom=Filled ):
[image]

futurau, 2021-09-21:
[image]

svenhb, 2021-10-04:
Check new release 1.6.2.0 for automatic fiducial correction:
https://grbl-plotter.de/index.php?id=fiducial-correction&setlang=en

svenhb, 2021-11-01:
No feedback is a good feedback 😉
Next release will have option to discard fiducial G-Code

closed 2021-11-01 by svenhb