svenhb/GRBL-Plotter #227: Ascii nc file

SimonJohnCastle, 2021-09-26:
It looks like the exported nc gcode file is UTF-16 which adds an additionl step if I want to use it else where as I have to convert it to 8 bit ascii.
Is there a reason for this ?, is it a standard ?

Could you add an option for the saved output format.

Thanks for all you hard work on this plotter and making it available to all.

trantai369, 2021-09-27:
edit code "fCTBCode.SaveToFile(saveFileDialog1.FileName, Encoding.ASCII);"
Encoding.ASCII

svenhb, 2021-09-27:
Didn't know this could be a problem...
I choosed "UTF-8" for the next release (new option to select between UTF-16 / UTF-8)

SimonJohnCastle, 2021-09-28:
Thank you.
Could you have Ascii as an option so there is no BOM (Byte Order Mark) saved at the begining of the fille.

svenhb, 2021-10-04:
Check new option in release 1.6.2.0:
I hope it works - I didn't check function.
[image]

SimonJohnCastle, 2021-10-04:
works as expected, thank you

closed 2021-10-04 by SimonJohnCastle