svenhb/GRBL-Plotter #229: The large file will load but not run.

rlwoodjr, 2021-09-28:
**Describe the bug**
I have large gcode file that will not run once you click play. It works on other senders.

**To Reproduce**
Load a gcode file with 500,000 lines (XYZ) wait until loaded, then click play.  It looks like it is going to run, but only a few lines, then it stops, waits for a few seconds and starts.  I will not run the file and the screen locks until the next bit

**Expected behavior**
Wait for a minute or two to load, then run the file.

The large gcode file can be downloaded here:
https://cdn.shopify.com/s/files/1/1563/9873/files/V-Carve_LL.gcode?v=1632859331

Is there settings I need to change?

rlwoodjr, 2021-09-28:
I also get these lines in the com port window.

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Missing 12 Real-time Status Reports per 10 seconds

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Missing -21 Real-time Status Reports per 10 seconds

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Missing 42 Real-time Status Reports per 10 seconds

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Missing -34 Real-time Status Reports per 10 seconds

rlwoodjr, 2021-09-28:
When the graphic is zoomed out it runs good
[image]
.

rlwoodjr, 2021-09-28:
When the graphic is zoomed in , it stall
[image]
s.

svenhb, 2021-09-29:
If you create the code by graphics import, I recommend to use "Remove very short moves"  perhaps < 0.2 ...
[image]

Because your code shows really tiny step-width, which are probably not needed, but may causes a lot of trouble (I assume these are mm not inch). 
With high feedrate grbl sends many "ok" messages (because command line  was parsed and executed) per second and fill up the receive buffer of the pc.
[image]

svenhb, 2021-09-29:
>The large gcode file can be downloaded here 

Ok it's ready to use G-Code... Probably in inch. 441518 lines. 
No idea how to improove the behavior - loading with a notebook i5 8th gen. is ok. It's loaded and zoomable after 12 seconds.

svenhb, 2021-09-29:
About missing status reports, try to reduce polling frequency:
[image]

rlwoodjr, 2021-09-29:
The gcode file is written by VCarve Pro.  I do have the Status poll frequency set to 2 Hz.

If the graphics are zoomed out, then it runs fine.  It stalls when the graphics are larger.  I do not think it is grbl that is locked.  I can try running this on the ESP32 and see if there is a difference.

Is there a way to turn off or reduce the graphics calculations for a large file?

rlwoodjr, 2021-09-29:
> > The large gcode file can be downloaded here
> 
> Ok it's ready to use G-Code... Probably in inch. 441518 lines. No idea how to improove the behavior - loading with a notebook i5 8th gen. is ok. It's loaded and zoomable after 12 seconds.

Did you try to run the file, zoom in or Zoom out?

rlwoodjr, 2021-09-29:
One last question.  I have downloaded the source code and maybe I changed something while I was exploring the code..   Does the software write any configuration files in other directories other than the installed folder?

svenhb, 2021-09-29:
>Does the software write any configuration files in other directories other than the installed folder?

Yes, the configuration, stored via "Properties.Settings.Default.xyz" - somwhere User\...\AppData\Local\GRBL-Plotter\GRBL-Plotter.exe_Url_5v3rojcbm....

>Did you try to run the file, zoom in or Zoom out?

Not yet

>Is there a way to turn off or reduce the graphics calculations for a large file?

Perhaps this helps, if switched off:
[image]

rlwoodjr, 2021-09-29:
Checking this setting did the trick!  Working good.  Thanks
[image]
.

rlwoodjr, 2021-09-29:
Thanks for the quick response.

closed 2021-09-29 by rlwoodjr

svenhb, 2021-10-04:
New release 1.6.2.0
I improoved loading of large g-code files a bit, also take care of G20 (units = inch).
But strong zooming is still not good.