svenhb/GRBL-Plotter #230: Alarm/Door disable M7 (Mist) state

amoineau, 2021-09-30:
**Describe the bug**

It's probably an intended behaviour not a bug, but when triggering an alarm or door signal the M7 state is reset. 

**Expected behavior**

I wanted to know if it could be avoided, if we could keep the M7 state during emergency stop or at least get it back to the previous step when resuming?
On our machine, M7 is used to pilot a relay that switch between the two tools we have and opening the hood triggers a door signal.
So if someone pause mid file to check something and open the machine it then resume with the wrong tool, which ruin the entire file and potentially hours of process. 

Thank you in advance for your help.

svenhb, 2021-09-30:
If you enable this checkbox, you should see all sent commands: I'm sure there is no M9 to switch off M7 state. Probably it comes from grbl itself?
[image]

Perhaps someoe pressed stop and this code will be performed?:
[image]

amoineau, 2021-09-30:
I was afraid that it came from GRBL itself shutting down the accessories when the Door input is triggered... In which case it will be hard to solve.

I only have M5, no M9, in the "send following gcode...". So I don't think the problem is coming from there :/

svenhb, 2021-09-30:
If you check grbls protocol.c https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/protocol.c you find:
grbl will switch off spindle and cooling - makes sense in connection with cnc
```
      // Parking manager. Handles de/re-energizing, switch state checks, and parking motions for
      // the safety door and sleep states.
      if (sys.state & (STATE_SAFETY_DOOR | STATE_SLEEP)) {

        // Handles retraction motions and de-energizing.
        if (bit_isfalse(sys.suspend,SUSPEND_RETRACT_COMPLETE)) {

          // Ensure any prior spindle stop override is disabled at start of safety door routine.
          sys.spindle_stop_ovr = SPINDLE_STOP_OVR_DISABLED;

          #ifndef PARKING_ENABLE

            spindle_set_state(SPINDLE_DISABLE,0.0); // De-energize
            coolant_set_state(COOLANT_DISABLE);     // De-energize

```

amoineau, 2021-09-30:
How did you find that x)
You are amazing. So if I comment the last line, it shouldn't de-energize the M7 anymore ?

svenhb, 2021-09-30:
>How did you find that 

I was just interested (that GRBL-Plotter is not be the resson for that ;-) and had an idea where and what to look for.

>So if I comment the last line, it shouldn't de-energize the M7 anymore ?

I don't know - just try. Perhaps there is some other code which handles the coolant state

amoineau, 2021-09-30:
Sure i'll try it, it's just that the machine is now 400 km away from my office, so it's getting harder to test stuff like that ^^'

amoineau, 2021-10-13:
Oh no, I wasn't suggesting that your software was the problem, I watched the detailed "Real time status report" and it seems it only sends the resume command. I still believe that it might come from how GRBL handles the resume signal but I can't find where is resets the coolant.
Ok I'll follow that lead, thank you !

svenhb, 2021-10-13:
Also I found here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptionsascii%E5%AE%9E%E6%97%B6%E5%91%BD%E4%BB%A4%E8%AF%B4%E6%98%8E

**If the parking compile-time option is enabled and the safety door state is ready to resume, Grbl will re-enable the spindle and coolant, move back into position, and then resume.**

Did you enbale this compile option?

amoineau, 2021-10-13:
Ok thanks for all the help

svenhb, 2021-11-01:
I hope you could fix it. Because it's not GRBL-Plotter reason, I close this issue.

closed 2021-11-01 by svenhb

amoineau, 2021-11-02:
Hello, I didn't find a fix yet, but you are right, this is not your problem, thanks for the help.