svenhb/GRBL-Plotter #231: Positions dont update

yagizzha, 2021-09-30:
[image]

messages are an array of coordinates like "X205 Y488 ", when i start this function it gets stuck inside the while loop because X and Y positions dont update but it still goes to the given position . i've tried to make and use a public version of your ProcessGrblPositionUpdate function but it didnt help . 

Is there a way to send this function a signal when the machine reaches its destination or a way to update the X and Y values real time ?

svenhb, 2021-10-01:
Instead of
`if (xabs + yabs < 0.001)`
Which is wrong in my eyes, you could use
`if (Grbl.posChanged)`

svenhb, 2021-10-01:
Grbl.PosChanged will be cleared after usage - probably not save for you.
In next release you could use 
`if (Grbl.Status != GrblState.idle)`

svenhb, 2021-11-01:
No feedback is a good feedback 😉

closed 2021-11-01 by svenhb