svenhb/GRBL-Plotter #232: Missing files

yagizzha, 2021-10-04:
[image]

there are missing files in the new version

svenhb, 2021-10-04:
Before publishing, I renamed all files from the lowercase first letter to the uppercase, and the files are here: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/GCodeCreation

yagizzha, 2021-10-04:
there is Graphic**Related**Tangential.cs and Graphic**Related**HatchFill.cs  but when i changed those into the missing files i got this error 
[image]
thats is why i assumed there was a file missing , if those are the correct files , what is cbroundup meant to be as in type , so i can add it to the class

svenhb, 2021-10-04:
Did you also update this folder: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/MachineControl
I renamed from gcode... to Gcode...

yagizzha, 2021-10-04:
i have downloaded full source code from here : https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.2.0 

every GCode is in upper case 
[image]

svenhb, 2021-10-04:
CbRoundUp is used in ControlHeightMap.cs and ControlHeightMap.Designer.cs my files are from 30.08. no need to update again

yagizzha, 2021-10-04:
New designer file fixed it , thanks

closed 2021-10-04 by yagizzha