svenhb/GRBL-Plotter #233: Teach scaling problem

yagizzha, 2021-10-04:
[image]
After teaching scaling noticed that it was going near double the amount it was supposed to . Calibration was done same as last version .

svenhb, 2021-10-04:
You should select square or circle - not both.
[image]

How does your fiducials look like? I just can see a dot.
Which camera mounting do you use?
[image]

svenhb, 2021-10-04:
Did you follow the distortion correction? https://grbl-plotter.de/index.php?id=form-camera&setlang=en#movable-mounted
If you don't needed it, you need to teach them to the outermost edges of the image, because the correction is enabled by default after opening the camera form.

svenhb, 2021-11-01:
No feedback is a good feedback 😉

closed 2021-11-01 by svenhb

yagizzha, 2021-11-03:
> No feedback is a good feedback 😉

Sorry about no reply , completely slipped my mind , after your last update you basically implemented all we were working on so there was no need to continue for us 😄