svenhb/GRBL-Plotter #236: MKS DLC 2.0  (Lasergrlb)   put a servo to down and up the pencil

JulioIS, 2021-11-08:
Hi, can you help me?
 I have a small problem I have a cnc with diode laser and I would also like to use it with a pencil to draw with a card (MKS DLC 2.0) axes (x & y)
I would like to put a servo to down and up the pencil,    I don't know what configuration to put in the software (Lasergrlb) in this case. 

Thanks in advance

svenhb, 2021-11-08:
If you control the laser via PWM, you can't control the servo at the same time. 
You need an extra electronic to generate the servo pwm, controlled by an on/off signal.
I made this electronic: http://svenhb.bplaced.net/?Elektronik___Stiftheber to control a servo.
Also usd here:https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en
If you can get your Z axis working, better do it this way: https://grbl-plotter.de/index.php?id=laser-pen

svenhb, 2021-11-11:
Also if you use RC-servo specific grbl-version (https://grbl-plotter.de/index.php?id=quick-guide#pwm) you will not get suitable PWM duty cycle to control the laser between 0% and 100%. 
With the given 61Hz (=16 ms) and 0.5 to 2.5 ms pulse width, you get duty cycles from 3% to 15%.