svenhb/GRBL-Plotter #238: Clean GCode export

wiring80, 2021-11-28:
I use GRBL-Plotter with an MKS-DLC with an TFT24 for offline plotting.

So i create the GCode and export it to SD Card, but must clean the GCode file from all bracket "(" and ")" with its included content, with Notepad++ or equal program.

Its not really important, but would be nice if a export setting will be added, for clean GCode generation, cause the TFT24 stop streaming the GCode, if found rows with brackets etc.. No matter if the bracket found at start at the row, or after GCode.

Anyway, the MKS TFT24 is buggy in many ways :-/

svenhb, 2021-11-29:
Yes I can implement this feature.

svenhb, 2021-11-29:
Hm, I don't have so much choices for encoding:
System.Text.ASCIIEncoding
System.Text.UnicodeEncoding
System.Text.UTF32Encoding
System.Text.UTF7Encoding
System.Text.UTF8Encoding

wiring80, 2021-11-29:
Mhmm, I don't have my PC ready at the moment, cause my screen is defective and on the way to the seller for warranty. I use visual studio for my private developments too.

With c# that should be a way for converting Utf8 to ISO without - more to say filtered signs that are not available in ISO:

Encoding iso = Encoding.GetEncoding("ISO-8859-1");
Encoding utf8 = Encoding.UTF8;
byte[] utf_Bytes = utf8.GetBytes(text);
byte[] iso_Bytes = Encoding.Convert(utf8,iso,utf_Bytes);
string output = iso.GetString(iso_Bytes);

svenhb, 2021-11-29:
This works: Encoding.GetEncoding("ISO-8859-1");

wiring80, 2021-11-29:
Perfect :-)

Ah, denn Mal Grüße zurück aus Nähe Hagen NRW :-)

svenhb, 2021-11-30:
Try new version: https://github.com/svenhb/GRBL-Plotter/releases

wiring80, 2021-12-01:
Thanks a lot. Seems to work, however - Notepad++ shows the file as Utf8, but choosed Iso 8859-1 in GRBL Plotter. Curios thing - my Linux GEdit tells me the correct encoding, iso 8859-1. So, think GRBL Plotter does it right :-)

closed 2021-12-01 by wiring80