svenhb/GRBL-Plotter #239: Window scaling/DPI awareness problem

AndrasMa, 2021-12-03:
Starting the program, the window is scaling well on 4k (UHD, 3840*2160 px) displays, scaling set to 200% in display settings.
Upon loading an svg, the main window is reloading, and the scaling is not any more working, everything gets way too small to read. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start grbl-plotter.exe
2. Looks fine
3. Click File/Open File...
4. Load the bunny from examples
5. Window reloads, lost scaling, everything is small.

**Expected behavior**
Scaling (DPI awareness, etc...) should remain the same as initial starting the program.

**Screenshots**
[image]

[image]

svenhb, 2021-12-04:
I can also observe the behavior, but after doing your workarround it seems to work - as you wrote... 
I think it works for any further program start, problem solved - or not? 
Would be nice to set this option (Override high dpi scaling) during install...

AndrasMa, 2021-12-04:
What happens during loading of saved files? May be it just needs to apply some flags, and settings for the main window.

Before loading it works fine, therefore upon first start scaling is all set up well. I've programmed just a little in Visual Studio C"anything" , so my memories are fading, but upon creating the window you have to set these things, and seems to be right. 

May be the bug lies somewhere in UpdateWholeApplication() function in file GRBL-Plotter/GUI/MainFormUpdate.cs which is called when different INI files is selected.

jem-green, 2022-01-03:
Workaround solves part of the problem but the rest of the scaling then breaks
[image]

smakhtin, 2024-05-12:
I started the DPI PR [here](https://github.com/svenhb/GRBL-Plotter/pull/398).