svenhb/GRBL-Plotter #240: How to remove circles from "Image to Gcode" results?

averydelmiller, 2021-12-07:
Hello!
I've been experimenting with converting images into gcode, and I'm loving the results. However, on some images, there are a large amount of small circles around the outlines (with some even in the filled area), seen in the screenshot below.
[image]
Sometimes it's not a big deal, but with this particular image the amount of circles is redundant, as many of them overlap in addition to extending the plotting time.
Any advice on how to manage this?
Thanks.

svenhb, 2021-12-07:
To be able to understand it: can you upload the image and your settings (Export Machine Settings)?

svenhb, 2021-12-10:
Did you try to switch off "smooth" and "Shrink path" or the whole "Draw outline" ?
It seems that "Shrink path" may cause the effect...
[image]

svenhb, 2022-04-01:
No feedback - hopefully solved

closed 2022-04-01 by svenhb