svenhb/GRBL-Plotter #242: Error 24 on a seemingly normal g-code file

amoineau, 2021-12-14:
Hello again,

The process sometimes get stuck in the middle of a file :

[image]

The problem seems to come from the streaming, we have an "error 24" appearing. 
I was able to unstuck the process by playing around with "Pause/Start Streaming"  and a bit of "Resume".

Here are the logs for when it happened :

[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/7711967/logfile.txt)

It wasn't the first time processing this file, so it doesn't seem to be the problem, but I could send you the file by mail if you want.

Thank you for your help.

svenhb, 2021-12-14:
I think there are too many commands per second, which needs to be sent:
Feedrate = 10000 mm / min = 167 mm / sec
Distance from code line to code line ca. 3 mm
Means 55 code lines per second are needed to transmit
One code line consists of ca. 18 characters -> 990 character per second

I know, with regular 115200 Baud = ca. 11520 characters per second should be possible to transmit...

See also: https://github.com/gnea/grbl/issues/1036

amoineau, 2021-12-14:
Sorry, what's "ca." ? :/

svenhb, 2021-12-14:
Yes approx. would be better - sorry.

amoineau, 2021-12-15:
Ok, so : to avoid having no commands in the buffer, you have to send enough commands to allow moving at a speed of 167mm/sec. In our case this means 55 line per seconds. But you are telling me that it shouldn't be a problem with our baudrate ?

The fact that 99% of the time we are able to process the files might concurs with that statement, plus  we often go faster than 10.000mm/min and it still works. 

Can I ask you what made you think that the problem was related to that ?

svenhb, 2021-12-15:
I mean, the baud rate can't be the problem, but perhaps receiving and processing commands inside the controller / grbl.
To be sure you could compile grbl with an option which echos all received commands, perhaps it helps to find the cause.
In GRBL-Plotter you could watch all actually sent commands, but it slows down the whole streaming...

I don't know how to find where the cause is...

amoineau, 2021-12-15:
> but perhaps receiving and processing commands inside the controller / grbl.

The error 24 is coming from GRBL itself ? It's not something grbl plotter "raises" after running into a problem?

>To be sure you could compile grbl with an option which echos all received commands, perhaps it helps to find the cause.

Does it have any major drawbacks ? Like, can we let it on all the time, until we face the problem again, without creating other problems or slow downs ?

svenhb, 2021-12-15:
Error 24 - now with slightly different text:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

>Does it have any major drawbacks ? 
 
I don't know, you need to figure out. If I remember: it is not recommended to keep it switched on. You need to check all the *.h files (of grbl)

amoineau, 2021-12-16:
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages
Yes I found that after you last message, I'll see what I can do. Thank you.

svenhb, 2021-12-17:
About #define REPORT_ECHO_LINE_RECEIVED
Check https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/config.h line 438
I will build in a log-option to save the echo-results to a file - for later compare.
I hope regular file-write access is fast enough for up to 55 code-lines per second - as discussed above.

svenhb, 2021-12-17:
Check log-options to compare echoed commands with sent commands: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2021-12-20:
Ok thanks !

closed 2022-04-01 by svenhb