svenhb/GRBL-Plotter #244: Trouble getting tool changes to function

wunderaa, 2022-01-02:
I have entered custom tool change commands into the *.nc files for remove and select.  When I generate g code, these seemingly are skipped over.  Running the code by itself (and manually plugging in the TOAX and TOAY numbers), it works as expected.  I have tried running manual calls for Tx M6 to trigger it and monitoring the Rx/Tx dialog, but just not seeing where the program is parsing through the code file.  Any suggestions?

From Serial screen when i send a T1 M6 command:

[Start streaming - no echo]
[Save last pos.: X=  345.700 Y=  276.000 Z=   25.000]
[Tool T1 removed]
[Tool T1 selected]
[Tool change finished]
[MSG:Pgm End]
[Pgm End*]

[Streaming finish line:2]

Accessory State run times :
             h: m: s
Spindle: 00000:28:37
> $G
< ok
< ok
< ok
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F0 S100]
< ok

From Remove file:
G53 G90 G0 X#TOLX Y#TOLY	
G91 G0 Z23					
Y-4
X-18.5
Z-19
M3 S100
Y4
X18.5
Z-4
G90

From Select File:
G53 G90 G0 X#TOAX Y#TOAY	
M3 S100
G91 Z4
X-18.5							
Y-4
M3 S20
Z19
X18.5
Y4
Z-23
G90

svenhb, 2022-01-02:
I thing the problem is the relative filepath, try to put in the absolute path in the setup (copy the path-string manually from file-explorer, don't select via "..." button):
[image]

wunderaa, 2022-01-02:
Spot-on, thank you!  So glad it was something simple!

Awesome work and thank you again!

closed 2022-01-02 by wunderaa