svenhb/GRBL-Plotter #245: G-Code Transfrom

jem-green, 2022-01-03:
I have the following generated from a SVG test file - using plotter.ini

(<Figure Id="1" Geometry="rect" PenColor="000000" PenWidth="0.265" Layer="layer1-Layer 1" PathID="rect839" PathLength="128.0"> )
G00 X0.000 Y0.000  
G01 Z-2.000 F500 (PD)
G01 X32.000 Y0.000 F1000 
G01 X32.000 Y32.000  
G01 X0.000 Y32.000  
G01 X0.000 Y0.000  
G00 Z2.000 (PU)
(</Figure>)

I'm think im using the g-code transform correctly to replace the Z with a spindle speed but the output looks odd, so im guessing this is not the way to create g-code for 2.5d plotter. The manual pen up/down buttons work as expected.

M03 S0
M03 S0
G00 X0.000 Y0.000 S0
M03 S0
G01 X32.000 Y0.000 S0
G01 X32.000 Y32.000 S0
G01 X0.000 Y32.000 S0
G01 X0.000 Y0.000 S0
M03 S0
M03 S0
M03 S0

I want it to look like this

(<Figure Id="1" Geometry="rect" PenColor="000000" PenWidth="0.265" Layer="layer1-Layer 1" PathID="rect839" PathLength="128.0"> )
G00 X0.000 Y0.000  
M03 S30
G01 X32.000 Y0.000 F1000 
G01 X32.000 Y32.000  
G01 X0.000 Y32.000  
G01 X0.000 Y0.000  
M03 S40(PU)
(</Figure>)

svenhb, 2022-01-04:
If you already import from SVG, you should choose the S (PWM)-Setup from here:
[image]

Using from menu "G-Code Transform - Convert Z to S" is for already generated G-Code and can be setup in Setup:
[image]

### But anyway, there is a bug...

jem-green, 2022-01-04:
So the bug i have noted seems to apply to both at the SVG import stage and the modification. I can review the code if it can be agreed on what the output should look like, rather than what i think i want to see.

svenhb, 2022-01-04:
I fixed the bug in modification "G-Code Transform - Convert Z to S".
[image]

I don't see the bug in the import section, looks ok for me.
[image]

I will upload the fix later this day.

jem-green, 2022-01-04:
Ok might be my misunderstanding - I used the PWM default so matches the setting above.
I then did a SVG import using the example_plotter.ini
I get this as output

( Use case: example_plotter.ini )
( SVG import by GRBL-Plotter 1.6.3.5 )
( Source: C:\Users\Jeremy\OneDrive\Documents\drawing.svg )
(<Header >)
( G-Code lines: 36 )
( Pen Down/Up : 5 times )
( Duration ca.: 0.3 min. )
( Conv. time  : 00:00:00.0525091 )
( Original graphic dimension min:4.000;4.000  max:36.000;36.000)
(</Header >)
F1000 (Setup - GCode-Header)
G90
G00 Z2.000
M3 S1000 
(<Figure Id="1" Geometry="rect" PenColor="000000" PenWidth="0.265" Layer="layer1-Layer 1" PathID="rect839" PathLength="128.0"> )
G00 X0.000 Y0.000  
G01 Z-2.000 F500 (PD)
G01 X32.000 Y0.000 F1000 
G01 X32.000 Y32.000  
G01 X0.000 Y32.000  
G01 X0.000 Y0.000  
G00 Z2.000 (PU)
(</Figure>)
(<Figure Id="2" Geometry="m" PenColor="000000" PenWidth="0.265" Layer="layer1-Layer 1" PathID="path845" PathLength="5.7"> )
G00 X16.000 Y4.000  
G01 Z-2.000 F500 (PD)
G01 X16.000 Y20.000 F1000 
G01 X12.000 Y16.000  
G00 Z2.000 (PU)

I note that the settings have reverted
[image]

jem-green, 2022-01-04:
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="40mm"
   height="40mm"
   viewBox="0 0 40 40"
   version="1.1"
   id="svg8"
   inkscape:version="1.0.2-2 (e86c870879, 2021-01-15)"
   sodipodi:docname="drawing.svg">
  <defs
     id="defs2" />
  <sodipodi:namedview
     id="base"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:pageopacity="0.0"
     inkscape:pageshadow="2"
     inkscape:zoom="3.959798"
     inkscape:cx="0.78999709"
     inkscape:cy="78.876288"
     inkscape:document-units="mm"
     inkscape:current-layer="layer1"
     inkscape:document-rotation="0"
     showgrid="true"
     inkscape:pagecheckerboard="false"
     showguides="true"
     inkscape:window-width="1920"
     inkscape:window-height="992"
     inkscape:window-x="-8"
     inkscape:window-y="-8"
     inkscape:window-maximized="1">
    <inkscape:grid
       type="xygrid"
       id="grid837"
       units="mm"
       spacingx="0.99999997"
       spacingy="0.99999997"
       empspacing="4" />
  </sodipodi:namedview>
  <metadata
     id="metadata5">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1">
    <rect
       style="fill:none;stroke:#000000;stroke-width:0.264583;stroke-opacity:1"
       id="rect839"
       width="31.999998"
       height="31.999998"
       x="3.9999998"
       y="3.9999998" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.229136px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="M 15.999999,7.9999994 V 12 h 3 V 7.9999994"
       id="path841" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.214337px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="M 20.999999,15.000001 V 7.9999998 h 3 v 3.5000012 h -3"
       id="path843" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 19.999999,15.999999 -4,4"
       id="path845" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 23.999999,19.999999 -4,-4"
       id="path847" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 19.999999,31.999999 v -16"
       id="path849" />
  </g>
</svg>

jem-green, 2022-01-04:
`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="40mm"
   height="40mm"
   viewBox="0 0 40 40"
   version="1.1"
   id="svg8"
   inkscape:version="1.0.2-2 (e86c870879, 2021-01-15)"
   sodipodi:docname="drawing.svg">
  <defs
     id="defs2" />
  <sodipodi:namedview
     id="base"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:pageopacity="0.0"
     inkscape:pageshadow="2"
     inkscape:zoom="3.959798"
     inkscape:cx="0.78999709"
     inkscape:cy="78.876288"
     inkscape:document-units="mm"
     inkscape:current-layer="layer1"
     inkscape:document-rotation="0"
     showgrid="true"
     inkscape:pagecheckerboard="false"
     showguides="true"
     inkscape:window-width="1920"
     inkscape:window-height="992"
     inkscape:window-x="-8"
     inkscape:window-y="-8"
     inkscape:window-maximized="1">
    <inkscape:grid
       type="xygrid"
       id="grid837"
       units="mm"
       spacingx="0.99999997"
       spacingy="0.99999997"
       empspacing="4" />
  </sodipodi:namedview>
  <metadata
     id="metadata5">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1">
    <rect
       style="fill:none;stroke:#000000;stroke-width:0.264583;stroke-opacity:1"
       id="rect839"
       width="31.999998"
       height="31.999998"
       x="3.9999998"
       y="3.9999998" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.229136px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="M 15.999999,7.9999994 V 12 h 3 V 7.9999994"
       id="path841" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.214337px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="M 20.999999,15.000001 V 7.9999998 h 3 v 3.5000012 h -3"
       id="path843" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 19.999999,15.999999 -4,4"
       id="path845" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 23.999999,19.999999 -4,-4"
       id="path847" />
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="m 19.999999,31.999999 v -16"
       id="path849" />
  </g>
</svg>
`

svenhb, 2022-01-05:
Check latest release 1.6.4.0 for bug fix: https://github.com/svenhb/GRBL-Plotter/releases

jem-green, 2022-01-05:
missing a file from the solution - Severity	Code	Description	Project	File	Line	Suppression State
Error		Could not copy the file "C:\Users\Jeremy\OneDrive\Source\Projects\GRBL-Plotter\GRBL-Plotter\data\usecases\example_plotter_Z_axis.ini" because it was not found.	GRBL-Plotter			

is this important?

svenhb, 2022-01-05:
I just renamed the file you mentioned: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/data/usecases/example_plotter_Z_axis.ini

jem-green, 2022-01-05:
Apologies if this is not how the setup G-code generation  is intended to work.
Is it worth separating my observations into separate issues.

1) Importing a svg file (previously included in a comment)

1a) Using setup i select the PWM option
[image]
1b) after the import there is no replacement of the G0/G1 Z PU/PU commands
[image]
1c) The setup reverts to using Z axis

jem-green, 2022-01-05:
Ok i think im closer to what i might be doing wrong

1b) If i press the yellow button then i get the expected behaviour

[image]

[image]

1c) What is the green button supposed to do - apart from changing the Set-up settings. as observed above.

jem-green, 2022-01-05:
Think we are closed on the original issue - just need to understand the green and yellow button press actions.

svenhb, 2022-01-05:
>Think we are closed on the original issue - just need to understand the green and yellow button press actions. 

Any idea how to make it more clear on first usage? 
Perhaps as default after program installation, this use-case dialog should be disabled?

jem-green, 2022-01-05:
Again thanks for you patience. So as someone new to the software i completely missed the concept of a "use cases" which trying to expand this is the settings profile for the graphics import.

1) I assume that this only includes those settings referred to from that tab.
2) Had the dialog NOT popped up i would have naturally experimented with the settings and seen the effects on the generated g-code.
3) I think i need to understand why the dialog was added at this point
3a) This seems to be for users with multiple plotters / CNC/ etc.
3b) Was the intention that these settings should just apply to this import, and not do any resetting?
4) It looks like the application graphic import settings are saved on close
4a) so there is a concept of a default or current but this is not indicated
[image]

So suggestion is to:-
 default to not showing the dialog
 Add a new checkbox to the use cases tab to say show dialog at graphics import
or
 Add an indication that you are using the current settings detailed in the graphics import setup
Reword to say "Override current/loaded settings with selected use cases" - green
Reword "Apply current/loaded settings" - yellow

svenhb, 2022-01-07:
Thanks for the analysis.
1) Saving as use-case will save just the shown settings. But you could also use a saved machine setting [Menu - File - Export machine settings] and copy the file to the needed subfolder, to be available as use-case.
3) Yes, intention is for use with different machines but also for differnent application - e.g. usage of tooltable with tool-change (perhaps manually) or just single color without tool change. Perhaps these are too much possible uses...

I will rework the use-case dialog and disable it on first start.
Thanks

svenhb, 2022-01-18:
I hope the use-case dialog is now more clear,
please check new release: https://github.com/svenhb/GRBL-Plotter/releases

jem-green, 2022-01-18:
Yes logically works for me as a new user.
Tested importing a svg, and selecting the current settings.
Tested importing a svg after change the current settings from setup
Works consistently.

closed 2022-01-18 by jem-green