svenhb/GRBL-Plotter #248: Required New Function In Grbl Plotter

trkadam58, 2022-01-13:
Need New function in GRBL Plotter as attached video link

https://youtu.be/h4IQuHPnOe4

svenhb, 2022-01-13:
You mean text import and text-path modification? I think thats toooooo much work.
Did you ever try Inkscape (https://inkscape.org/de/)? Just do it there, when graphic looks fine, convert text into paths, then select all paths and copy to clipboard, then paste from clipboard into GRBL-Plotter (ctrl-V on 2D-view).

trkadam58, 2022-01-13:
But i need this function  in GRBL-Plotter only, u made for me required small changes, i will Donate every month 100usd this is just my small Donation, today i have done 50Eruo

svenhb, 2022-01-18:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

svenhb, 2022-01-19:
You need to enable grouping to select whole text: 
[image]

On the lower left edge you also get some infos:
[image]

trkadam58, 2022-01-20:
is it possible select whole line without pressing control key

svenhb, 2022-04-01:
No feedback, seems to work :-)

closed 2022-04-01 by svenhb