svenhb/GRBL-Plotter #249: Question about installation method and performances

amoineau, 2022-01-14:
Hi,

I wanted to know if there could be differences concerning performances depending on how grbl plotter was installed.
I'm thinking "Setup" vs "zip" vs code "source executed via Visual Studio". Do you have any ideas on the topic ?

Thank you in advance for your time.

svenhb, 2022-01-14:
There should be no difference between using the setup or unzipped program-folder, except the path to the data-folder:
During setup the data-path will be generated and set in registry (check the file installer.iss).
GRBL-Plotter checks the registry for actual data-path.

If the setup was never used and the reg-key doesn't exists, the data-folder will be assumed in the application path - where the exe was started.

This reg-key is also read and applied if program started from unzipped folder or via visual-studio...
Check the "About" window for current data-folder path.

Starting the program from visual studio will slow down everything - my experience.

amoineau, 2022-01-14:
Ok thank you for all these information !

Considering that I change version after each of your release, does using the setup every time risk messing things up ? Should I uninstall before installing the newest version ?

For now I have been mostly using the zipped version just to be extra safe. That way I have a folder containing subfolder with every version of grbl plotter.

svenhb, 2022-01-14:
I don't think an uninstall is needed (perhaps depends on user right? Admin or normal user?)
The setup just replaces the updated files (if a there is already an installation on same place) and asks for replacement of changed data-files.

>For now I have been mostly using the zipped version just to be extra safe. That way I have a folder containing subfolder with every version of grbl plotter.

Probably the best solution to figure out where an error has crept in...

amoineau, 2022-01-14:
Ok good to know.

> Probably the best solution to figure out where an error has crept in...

Yes, it came in handy once or twice the past year and a half ^^

closed 2022-01-14 by amoineau