svenhb/GRBL-Plotter #251: Tool change script not found

amoineau, 2022-01-17:
**Disclaimer : This is not exactly a bug.**

**Description**
I'm using the tool change feature but without scripts, so my setup is empty :
[image]

Before version 1.6.4.0 I had no problem with that.
After switching to 1.6.4.0, I get hundreds of lines saying : "tool change script not found" in the console when starting streaming, and 5 time that in the logs.

**Expected behavior**
I wish there was a less worrying way of telling us we are not using a script, like a one liner.

Thank you for your help.

amoineau, 2022-01-17:
This is worsened by another problem I'd like to report : the layers tend to be split so that every shape has its own layer, even when using only one layer for the whole file. I made sure that the name in the dxf and the tool table are the same.

I believe this is why I get the error so many times.

svenhb, 2022-01-17:
I thing the problem is the relative filepath, try to put in the absolute path in the setup (copy the path-string manually from file-explorer, don't select via "..." button):
[image]

amoineau, 2022-01-17:
Ok I'll try that before the next file, thanks !

amoineau, 2022-01-17:
It did not solve the problem :/

svenhb, 2022-01-17:
Because of this issue https://github.com/svenhb/GRBL-Plotter/issues/244 I added the warnings...

I don't think you need to enable "Perform tool change" if you don't use any scripts - can you check?

svenhb, 2022-01-17:
I will change the behavior: if textbox is empty - no check if file is existing...

amoineau, 2022-01-17:
I'll check again, but I tried a long time ago and I think it didn't apply the gcode in the tool table if unchecked.

amoineau, 2022-01-17:
Ok, it seems to work with "Perform tool change" unchecked but I get this message when reaching a tool change :
```!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'T01M06'  line-Nr:51
Try to continue err-count:1```

svenhb, 2022-01-17:
Ok, I will take care

svenhb, 2022-01-18:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2022-01-19:
Cool thanks ! I should be able to try it today or tomorrow.

amoineau, 2022-01-26:
Sorry I forgot to give you a feedback ! It works well

svenhb, 2022-01-26:
Thanks for feedback

closed 2022-01-26 by svenhb

amoineau, 2022-01-27:
It's the least I can do ! Thank you for all the updates