svenhb/GRBL-Plotter #252: Additional control of A and B stepper motors

hamppi, 2022-01-24:
Hi, Sven

Wonder if one thing I need is already existing in grbl-plotter in some way or if its not I'd like to request it for some future development.

So the machine I'm building now require use of additional stepper motor (like letter A named) this stepper is used to feed ink on the paper. So I need something like additional A motor start after /pen down/ command finished to all of the commands that go after depending on the path length (it is similar to what 3d printers have with E extruder for plastic feed) and some control of feed would be also great.

I know subroutine can be used for something really close to that but I would get it in order and there is much need it would run at the same time as it goes through the line. Short example with subroutine ill get a stop every time its called:
G01 X21 Y27 Z5
(subroutine)
G91 A10

And I need it this way:
G01 X21 Y27 Z5 A10

Where A10 the 10 is connected to the travel distance and can be multiplied or divided by a user number may be overriding F speed would cool for those. Also a separate B control feed would be interesting for future too:
G01 X217 Y27 Z5 A10 B20

The only thing I found is adding tangential axis command but its for a different case so I don't really get if I can make it work for what I need. Is there other way I missed? I can give more information of what I mean if its hard to understand.

Thanks much for improving grbl-plotter its a really cool software already with many ways of use.

svenhb, 2022-01-24:
Hi, your idea sounds good to me. I built a similar hardware (syringe) but never used it (http://svenhb.bplaced.net/?CNC___Plotter_2).
Which options are needed to calculate the feed rate? Just the length of the current move command or could be some other parameters be interesting?

svenhb, 2022-01-24:
>Where A10 the 10 is connected to the travel distance and can be multiplied or divided by a user number may be overriding F speed would cool for those. Also a separate B control feed would be interesting for future too:
G01 X217 Y27 Z5 A10 B20

Just to think about:
- To get simultaneous motion, all axis commands needs to stay in one command line - a given feed rate counts for all axes...
- I don't think the travel distance is needed for calculation: you need x ml/distance of ink. I think it's directly  connected with the feed-rate (times a factor)...

What do you think?

hamppi, 2022-01-24:
> I built a similar hardware (syringe) but never used it (http://svenhb.bplaced.net/?CNC___Plotter_2).
Yes, I saw it and thought it might already be in grbl-plotter somewhere adding additional steppers. I was also trying to use linear bearing with syringe for ink but now switched and checking a simple waterpump stepper header, the air pressure from it is enough for ink injection and setup is way simplier this way. 

Well some calculations might be cool sure. Ive made some sketch on it, hope it can be useful and make it easy to understand what I mean and you can tell if its possible and easy to implement. 

[image]

One is a musthave - its a feed rate multiplicator to increase/decrease feedrate.

Another one is connected to use of Z axis couse at least for my use scenarios (I can imagine there might be plenty of them not connected to Z at all with calculating the feed rate) 

But Im using it for brush ink painting mostly and Z value is somewhat controls line thickness in my case. So with a nozzle paint Z converted to feedrate will increase the amount of ink ejected for example, and it is a way to make line more dynamic. (I can for example decrease my Z values on creating in rhino and then use feed rate multiplicator to increase amount of ink with having just really small and unnoticeable Z steps and many other options to control feed)

2nd scenario is for the brush with ink container and this way some increasing of feedrate based on Z value is also good because when brush goes deep into paper it leaves way more ink then when it just touches with the end so it need more of refill. 

Im not really sure if what I mean on "Inverted" in sketch can work like I think, but may be just tired and its okey.
Ready to answer sure to any questions on what I wrote here :)
 
Some video of what I mean with brush and Z.

https://user-images.githubusercontent.com/31763713/150783371-072113f1-d7e0-4567-a175-f4fb26ba61c0.mp4

https://user-images.githubusercontent.com/31763713/150783477-7afb2ae0-b1f4-4e85-9501-7244fbb12434.mp4

hamppi, 2022-01-24:
> Just to think about:
>
> * To get simultaneous motion, all axis commands needs to stay in one command line - a given feed rate counts for all axes...
> * I don't think the travel distance is needed for [...]

Was just writing this big message when got yours. In my mind its somehow directly connected to distance, something like when you spend ink on paper - there is need on refill otherwise where else would the ink go? Or you mean only the calculation think?

svenhb, 2022-01-24:
I think the problem would be the feed rates which can't be seperated - example:
**Setup:** Syringe connected to stepper, delivers 1 ml of ink / mm ($103= your setup)
**Assumption of needed ink:** 1ml / 10 mm distance
You want to move 50 mm with F600 (600 mm/min = 10 mm/sec) - takes 5 sec.
50 mm needs 5 ml of ink, within 5 seconds, which results to a feed rate of  5 ml (=mm) / sec. = 300....
If you set the GCode like this: `G1 X50 F600 A5 F300` it won't work, I would say...

svenhb, 2022-01-24:
In my eyes the only workaround is to set up the $103 value for the syringe A axis

svenhb, 2022-01-24:
Perhaps I wrong: I would calculate the needed A-"distance" for the given feed-rate of XYZ

hamppi, 2022-01-24:
> Perhaps I wrong: I would calculate the needed A-"distance" for the given feed-rate of XYZ

Ok so you mean distance with speed dependency? Sounds even better in my opinion.

svenhb, 2022-01-24:
I think finally there is no speed dependency: for a given distance, you need a specific amount of ink, thats it.
grbl will control the speed for all given axes to reach this goal...
GRBL-Plotter would add the A-axis with value "distance inside this command line" times a factor.

svenhb, 2022-01-24:
>Some video of what I mean with brush and Z.

Thanks for the nice videos.
There is an option "Pen width to Z" . 
If you import images, than "gray to hight" is also an option - in this video not for Z but for S, to control a servo: https://youtu.be/7Nvf4UwK_yk

hamppi, 2022-01-24:
But that's the thing, I'm already importing dxf with Z values and they are totally fine with me. I only need additional A B stepper with some options may be if its not too hard to make, like distance * multiplier and some Z options if its possible. Otherwise I sure tried already Pen width to Z and also was long fighting with servo but servo is horrible for ink injections unfortunately even with features of grb-plotter.

svenhb, 2022-01-25:
I made a 1st implementation and wondering if A,B,C uses the same motion mode as X,Y,Z - absolute or incremental.
And now you need a 2nd multiplication by Z-value with the listed options?
[image]

Absolute:
[image]

Incremental:
[image]

hamppi, 2022-01-25:
> I made a 1st implementation and wondering if A,B,C uses the same motion mode as X,Y,Z - absolute or incremental. And now you need a 2nd multiplication by Z-value with the listed options? [image]
> 
> Absolute: [image]
> 
> Incremental: [image]

Looks great!
As I see with Incremental in your code it will be going back and force so its not working for the ink injection and with absolute it looks like it will work  (thought might be something for other use scenarios so may be a button to switch it would be good for future use?)
But Z calculation as I think should be based on incremental number not on the absolute number, if you get me right here. And after calculation it will create an absolute one.

svenhb, 2022-01-25:
I think also, the absolute-mode should be default and incremental an option - how does it work on 3D-printers for the ABS-feeder?
For Z-depending B you mean: B-value = travel-distance (of this specific code-line) * Z-value * factor?

hamppi, 2022-01-25:
Well both ways exist and have their specific. If we want to have incremental extrusion like in some 3d printers there is need for if we use absolute for all axis:
G92 A0
after every time the extrusion code was used inclemently? 

Or is it possible to use M82 M83 to set the mode on A and B axis absolutely and inclemently? In 3d printers as I get its controlled by M82 and M83 software and printer dependent for the E axis.

[image]
[image]

This article seems to be good on this.
https://www.cnckitchen.com/blog/g-code-basics-for-3d-printing

The good thing about absolute is some ability to see the amount of ink for the project straight in the gcode may be, the bad thing is some difficulties with reading code may be ... Or what you say on this?

>For Z-depending B you mean: B-value = travel-distance (of this specific code-line) * Z-value * factor?
Yes, at least I think it should work correctly

svenhb, 2022-01-25:
I keep the option "sum up" but rename it, perhaps with "sum up" a "G10 L20 P0 A0" should be included in the start sequence, to reset the axis - as I would do on the GUI "Zero A".

svenhb, 2022-01-26:
Hope it works as expected, would be nice to see results: https://github.com/svenhb/GRBL-Plotter/releases

hamppi, 2022-01-27:
Awesome! Thanks a lot, it was amazingly fast.  
Going to check and comeback with some results.

hamppi, 2022-02-10:
From the first experiments it looks to be exactly like it should be with much variety of using! Going to make more tests still and didn't try two steppers at same time (2nd pump is on the way). But I didn't notice any calculation mistakes or so.

And even the idea of distance = ink use, seems to be fine. Also tried with high viscosity acryl and there is some delay between the stepper move and the ink goes on paper, but may be its more of hardware problem than the software should take care of.

Also got Idea that it might be interesting to have ability connecting pen width/layer style to the stepper speed somehow, don't think I personally can use atm it but sounds reasonable and I can see some cases for that (I just dont use pen width much but one that use it can have more ink extruded depending on the pen width or layer. On the other hand I think its possible to make it in the Tool table and selecting different Z values for colors ... dont have much experience with that) 

Long story short - going to continue checking and testing it more, but seems to be great!

Thanks a lot, Sven! Great work and fast implementing!

svenhb, 2022-02-10:
Thank you for the detailed feedback.

svenhb, 2022-04-01:
Seems still to work as expected :-)

closed 2022-04-01 by svenhb

hamppi, 2022-04-01:
Hi, Sven

I made just some bit more experiments and haven't found still any errors thought cant say I really made a lot on it last days. But hoping to continue asap. If there will be anything Ill create a report. 

Thank you!