svenhb/GRBL-Plotter #253: Not working IP function

trkadam58, 2022-01-26:
I have tested IP function with GRBL Advanced it is connecting but GRBL-PLOTTER getting Stuck, also i have tested with IP testing software, same problem facing here also, just for your trial i have added IP testing software link.

https://drive.google.com/file/d/1CcnsDOCXZlRIqeOHTIP1pyxAEbOiQzXq/view?usp=sharing

https://www.youtube.com/watch?v=8Zh50QU39x8

svenhb, 2022-01-26:
Yes it's connected to Ethernet but not to grbl: I don't see the reset-message, like this:
[image]

trkadam58, 2022-01-26:
I have check with GRBL i am getting error as below video Link and image.

https://youtu.be/lUIChDd8JkY
[image]
[image]

svenhb, 2022-01-26:
Which controller and firmware are you using?
As written, I tried with ESP32: https://github.com/bdring/Grbl_Esp32
On an ESP32 controller, which uses Telnet

trkadam58, 2022-01-26:
i am using STM32F446RE in LQFP64 package with GRBL-Advanced firmware

https://github.com/Schildkroet/GRBL-Advanced

svenhb, 2022-01-26:
Did you follow this: For ETHERNET an additional W5500 Module is required. Then uncomment ETH_IF in Platform.h.?
Does it work with Candle? https://github.com/Schildkroet/Candle2

trkadam58, 2022-01-26:
i have done required setting for ETHERNET also added W5500 Module, and uncomment ETH_IF in Platform.h, and it is working fine with candle software without any problem
[image]
.

svenhb, 2022-01-26:
Ok, good to know.
Which output do you get, if you enable "Show report" before connecting and then connect to ethernet?
[image]

trkadam58, 2022-01-27:
Before connection i have selected show real time status

[image]

After connection i have selected show real time status, there is no change.
[image]