svenhb/GRBL-Plotter #255: Circles turned into lines on DXF import

amoineau, 2022-02-08:
### Description
Hello Sven, 
We met a problem where small circles where replaced by a horizontal line the size of the expected diameter.
We were using a very big DXF file, with a lot of these circles (see below).
The alteration was visible in the g-code and the view window, so it surely happened during the import from dxf to gcode.

### To Reproduce
Open this file in GRBL-plotter :
[250x4-3 cloisons 1 laser.zip](https://github.com/svenhb/GRBL-Plotter/files/8022148/250x4-3.cloisons.1.laser.zip)

### Expected behavior
There should be a lot of 1 mm diameter circles inside the sail.

Thank you for your help !

svenhb, 2022-02-08:
This circles are designed as "DXFPolyline" in your file - during import I don't care about bulges.
I opened this file in LibreCad and saved a part of it, then this circles are saved as DXFLWPolyline - during import, here I take care of bulges and cirlcles will be displayed.
I will try to implement also bulge-import for DXFPolyline.
Also I noticed "error 33" messages - problem: too large numbers. During export I use float and loose accuracy.
Also zooming should be extended...

svenhb, 2022-02-08:
Check new version: https://github.com/svenhb/GRBL-Plotter/releases

amoineau, 2022-02-09:
Ok thanks a lot I will try that !

amoineau, 2022-02-14:
Ok so, the new version seems to be broken... A few lines after starting streaming the CNC Com get closed so everything stops. We tried resetting everything, closing and reopening the software, restarting the PC but we always get the same result.
I rolled back to the previous version and everything worked, so it doesn't seem to be a physical or board-related problem.

A few weird things I quickly noticed is that the message saying that 'script were missing' is back and the black and pink cursors are huge now. Probably not helpful though.

svenhb, 2022-02-14:
Can you provide a logfile?

>I quickly noticed is that the message saying that 'script were missing' is back

Perhaps new installation filled the edit-boxes again? They should be empty, if no script is selected:
[image]

>and the black and pink cursors are huge now.

With your example file, the markers were not visible anymore if completly zoomed in, I changed their size related to 2D-View size, not graphic size - as before. So the size stays the same, when zooming in or out.
Perhaps I should make the size adjustable

amoineau, 2022-02-14:
My bad for the scripts, I removed the second part but let the "data\scripts\". I forgot I had to remove everything.
Ok for the cursor.
I'll get the logs tomorrow morning.

amoineau, 2022-02-15:
Ok I think this one shows the problem :

[log_20220214.5.txt](https://github.com/svenhb/GRBL-Plotter/files/8068078/log_20220214.5.txt)

There are all the errors with the subroutine but I think it's because I didn't fully removed the scripts.
And then there is an exception with the serial port.

svenhb, 2022-02-15:
This is probably one disadvantage of the extended logging, the error says:
`System.IO.IOException: The process cannot access the file '...\logfiles\logSendBuffer.nc' because it is being used by another process.`

amoineau, 2022-02-15:
It's weird because nothing was open that could access this file, we even restarted the PC...
And we could simply close 1.6.5.2 and open 1.6.5.0 and it would work. Did anything related to that changed between the versions ?

svenhb, 2022-04-01:
Is hopefully solved here: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.6.0

closed 2022-04-01 by svenhb