svenhb/GRBL-Plotter #256: 4-axis function

lfmmd, 2022-02-13:
hello i have an idea
My application is to make a curved cutting tool using a 2mm width saw blade that needs to keep the cutting direction and the feeding direction consistent

svenhb, 2022-02-13:
Hello, I think the tangential function is, what you are looking for: You could also use the Z-axis as the tangential axis, which I've tried here with less success (due to lack of assembly rigidity): https://youtu.be/uMFyTCK8Ggw
[image]

lfmmd, 2022-02-13:
> 您好，我认为切向功能是您要查找的内容：您也可以使用 Z 轴作为切向轴，我在这里尝试过但不太成功（由于缺乏装配刚性）：[https:// youtu.be/uMFyTCK8Ggw](https://youtu.be/uMFyTCK8Ggw) [image]
Thank you so much for your reply yes this is exactly what I imagined

lfmmd, 2022-02-13:
> Hello, I think the tangential function is, what you are looking for: You could also use the Z-axis as the tangential axis, which I've tried here with less success (due to lack of assembly rigidity): https://youtu.be/uMFyTCK8Ggw [image]

Thank you so much for your reply yes this is exactly what I imagined

closed 2022-02-13 by lfmmd