svenhb/GRBL-Plotter #258: I get an error when I import Gcode files from FlatCam

quangbinhhvac, 2022-02-17:
Hi svenhb,
When I import files from flatcam software, I get an error. The same file is normal when using candle. Error like picture attached file

https://drive.google.com/open?id=1tEvsVXhRZAlKU5EZXr5bJec4Tdy4Sgg8&authuser=binh.mydata%40gmail.com&usp=drive_fs
https://drive.google.com/file/d/1tIoeaor7L3-QRF7qaKv1xTcQvHMQ2irf/view?usp=sharing

Also, where can I find the manual for creating elevation maps with PCB milling

Thank you very much!

svenhb, 2022-02-17:
Hello, I assume with error you mean the wide pen-width. You can adjust the pen-width for the 2D-view here, and you may switch off "Color mode" (Otherwise the pen-width will be determined from actual tool-table, combining given parameters in your G-Code):
[image]

Unfortunatly I don't have PCB specific manual, I hope this helps:
https://github.com/svenhb/GRBL-Plotter/wiki/surface-scan
https://github.com/svenhb/GRBL-Plotter/wiki/auto-leveling

quangbinhhvac, 2022-02-17:
Hi,
I did it by turning off "color mode". May I ask, is it just a change of display? and the milling code commands are unchanged
For the PCB I want to find the elevation map using the milling tip directly on the workpiece. The docs only say that a switch-type device must be connected to pin A5 and that the Z axis still moves an extra distance when the switch is closed. If I use it with a milling cutter and a copper PCB, it breaks through at the point of contact

svenhb, 2022-02-17:
Yes, turning off "color mode" just changes the 2D-view, nothing else.
>that the Z axis still moves an extra distance when the switch is closed

Yes because of deceleration of the motor, but the exact point of contact will be taken
>If I use it with a milling cutter and a copper PCB, it breaks through at the point of contact

You need to use low speeds for Z movement, to have a very short deceleration distance.
I'm sure other leveling tools are working in the same way...
You may look for videos "grbl pcb autoleveling", with GRBL-Plotter it should work in the same way, but I didn't try for a very long time.

quangbinhhvac, 2022-02-17:
I very much hope you will have an example video of this in the future
Thank you very much!

svenhb, 2022-02-17:
Yes I should make one.
Further feedback from you would be nice.
Thank you

svenhb, 2022-04-01:
I think problem was solved

closed 2022-04-01 by svenhb