svenhb/GRBL-Plotter #259: Pen (up/down) direction reversed for

autofrank, 2022-03-06:
Hiu
I have a standard setup and been through all the wiki and can't find a solution. Apologies if this is a simple one
In in the pen up/down commands (custom button) are acting in reverse. I tried adjust the servo direction in grbl but no impact 
If i run one of the sample svg (grbl_graphic_popup) it all works except the pen is up when it should be done (and visa vesa) 
I am running from the tool table and tried adjusting the settings for pen but couldn't see any chance but TBH,I don't really know what all the settings do (current settings attached )  
Is there a way to reverse the operation of  the pen

Thanks
.
[image]
frank

svenhb, 2022-03-06:
Hi,
you need to find the correct pwm-values for pen up and down and apply here:
If you enable "Send command to check" you can directly apply a new value, when changed in the control...
[image]

svenhb, 2022-03-06:
This two values will be taken for the custom buttons as variables #GMAS (Graphic MAx S) or #GMIS (Graphic MIn S)
[image]

autofrank, 2022-03-06:
> This two values will be taken for the custom buttons as variables #GMAS (Graphic MAx S) or #GMIS (Graphic MIn S) [image]

Thanks - Any tips on finding the correct PWM values or is it just a case of incrementing by 1 and testing each possibility ?

svenhb, 2022-03-06:
It depends on your grbl-setup: $30 and $31 settings are defining the max and min possible values - default is 1000 for max (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#---view-grbl-settings). 
It also depends on your mechanical setup...
Here you can see the how I did it at 1:20 minute - note, that I set the max pwm to 32: https://youtu.be/7Nvf4UwK_yk?t=80

autofrank, 2022-03-06:
> It depends on your grbl-setup: $30 and $31 settings are defining the max and min possible values - default is 1000 for max [...]

Thanks for the information

I got it working with the Individual settings 
[image]

but now it's loosing serial comms during a drawing
[image]

I'll take a look at what you sent on in case my approach is causing the comms dropout

svenhb, 2022-03-06:
So you need to set the Z axis? Strange, how is your servo connected? Because your wrote "I tried adjust the servo direction"...
Then you can use the normal Z axis setup...
About loosing connection - There may be various reasons: weak power supply, bad shieled USB cable, ....
What is the result of "Check GRBL" ?:
[image]

[image]

autofrank, 2022-03-06:
> So you need to set the Z axis? Strange, how is your servo connected? Because your wrote "I tried adjust the servo direction"...
> Then you can use the normal Z axis setup...

I'm a little new to this so still learning and may be doing something stupid
TBH - I dont understand the difference between the use of z-axis, PWM, Spindle and Individual so I've been just trying options.
When I had the pen going in wrong direction, I was looking in GRBL .h files as one had servo direction as an option but that was before I found your solution

I'm using a cnc arduino shield (red pcb) with a genuine arduino uno. Servo is connected to 5v and gnd and data to Z+
[image]

I tried the approach you had in the video but no value triggered any pen movement

autofrank, 2022-03-06:
> About loosing connection - There may be various reasons: weak power supply, bad shieled USB cable, ....
> What is the result of "Check GRBL" ?:

I'm using a 10A bench power supply and the cable that came with the shield but will try another 
nothing in "check GRBL" when it disconnects
[image]
 if I "open" comms" again, I just get the settings like

autofrank, 2022-03-06:
i switched back to using zaxis and it seem to be working (correct pen position) 
Swapped the cable and still comms are stopping (different places in the drawing , so no real pattern) 
[image]

autofrank, 2022-03-06:
Re the comms drop, I just noticed that one of the stepper motors was running v hot (even though the driver was not and PSU only drawing 800w for two nema17 stepper and boards ) but would like to check that out first 

Thanks

svenhb, 2022-03-07:
You must have a special grbl-version on your arduino: the driver for the z axis is not installed, but you need to send z commands to activate your servo...
With "Check grbl" the program reads the $-settings and chekcs if the numbers are ok. E.g. a number in not ok, if you set a high number of step/mm at $100 and also allows high speeds at $110, if you then run high speeds, the arduino is not fast enough to generate the step-pulses and a uncertain behavior appears - I run into this trap and had connection errors...

autofrank, 2022-03-07:
Thanks SvenI think I am running the eggbot version of grbl.Which version do you recommend I use for your grbl-plotter program.  - is it the servo version on the help pages ? Sent from my Galaxy

svenhb, 2022-03-07:
Yes, I would recommend the version I know (grbl_v1.1f_Servo.hex).
On the other hand, now you figured out how your version works - just keep the check-mark for the z-axis. 
From which location did you download your version? I wan't to check if there are some other pit falls...

svenhb, 2022-03-07:
Also hex-file can be uploaded via arduino ide (but looks a bit complicated).
You can also download the source and compile by your own: https://github.com/cprezzi/grbl-servo

autofrank, 2022-03-07:
> From which location did you download your version? I wan't to check if there are some other pit falls...
I couldn't find the repository but it is called "grbl-servo-eggbot"
I have a copy of it if you are interested

autofrank, 2022-03-07:
@svenhb 
I started from scratch  and took the grbl code from grbl-servo and uploaded it to the Uno
No errors on upload

I uninstalled  the version of grbl-plotter I had and I downloaded the latest version of Grbl-plotter Version 1.6.5.2 from https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.5.2

Could not get PWM servo to work 
Resorted to using individual commands

[image]

Pen heading in opposite direction again 
I could mirror the mechanism and print it again in reverse and this would solve the pen direction issue but I noticed something odd in g-code 

Line 55 - This is gcode for lower pen (but the correct action here should be raise pen) 
Line 56 - move to new location
Line 57 -This is gcode for raise pen (but the correct action here should be to lower pen)
Line 58 onwwards gcode for drawing the lines 

So it looks like the gcode is generated with the reverse - not sure what governs the raise v lower

I could revers the mechanism and it would but all would be in reverse..

Any thoughts  ?
 
[image]

svenhb, 2022-03-08:
>Pen heading in opposite direction again

If would have a real Z axis, I would say: You need to adapt $3 and invert the direction: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings

>Line 55 - This is gcode for lower pen (but the correct action here should be raise pen)

Did you try to set "-2" for  pen up and "2" for pen down for individual commands?

>My Z Axis motor is getting hot 

I'm confused... Do you have servo or a real Z axis? Perhaps you can make a picture of your machine?

>Re the comms fails

Perhaps you can send the logfile? Location: check the About window from the menu, where the Data path is located.
Thanks

autofrank, 2022-03-08:
This is my setup. Two Nema 17 stepper motors (X and Y ) , CNC shield and
DRV8255 drivers. - Servo for Z axis (

Did you try to set "-2" for pen up and "2" for pen down for individual
commands?
Tried that but no difference

Do you have servo or a real Z axis? Perhaps you can make a picture of your
machine?
maybe I should change to a stepper for Z as well ?, I have a small Nema 17
I could use - might be closer to my 3d prnter
need to design a new z mount but should be okay

Getting Hot - It's like as if the stepper is "station keeping " all the
time  and hence getting hot
I could try and ;lower the current theshhold to may 33% of max (maybe
around 0.5 volts ) and see if it improves the heat and maintains the
required torque
I could also swap the drivers and see if it stays with the motor

I'll check for the log file later and send on

svenhb, 2022-03-08:
BTW: a regular RC servo draws a lot of current, when moving. I also had problems because of that and added a seperat power supply for the servo (a 5V regulator 7805, supplied from the 12V power supply).

autofrank, 2022-03-08:
> I'll check for the log file later and send on
[log_20220307.6.txt](https://github.com/svenhb/GRBL-Plotter/files/8208461/log_20220307.6.txt)
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/8208462/logfile.txt)

svenhb, 2022-04-01:
Strange behavior of your firmware, but workarround was possible. It's not a bug in GRBL-Plotter.

closed 2022-04-01 by svenhb