svenhb/GRBL-Plotter #263: projector as real screen

Vincenzoferrara, 2022-03-20:
hi, I work in the field of leather goods, in my sector there are cnc that mount a projector (above) that shows where the design is positioned by projecting it into the working area.
I was wondering if such an implementation was possible.
perhaps to make it work you could use the projector as the second screen and create a formula that, based on the resolution and the distance, calculates the width it must have in pixels (or give a range of distances) I know it's a very specific case but I think be a nice implementation thanks in advance

ps: sorry for english but i am using google translate

[image]
[image]

svenhb, 2022-03-20:
Sounds interesting. I found this https://www.lap-laser.com/products/projection-system-configuration/
and see that this system needs its own special software...
Or do you mean a "regular" projector as used fo TV?

Vincenzoferrara, 2022-03-22:
I mean a normal projector, the one you use as a TV.
From some research I have found that it is not possible to turn off screen pixels, (but you may already know that). I will use a cheap one
so the background color and the color of the lines should be editable (since it works with the light on, use the color that creates the most contrast)

svenhb, 2022-03-23:
Then it may works in this way: displaying the 2D-view (or part of it) in a new form, which can be expanded to full screen. 
The projector is attached as an additional monitor and you just drag the new form to the projectors-desktop...
Somehow a teaching is needed to scale the drawing in the new form...

Vincenzoferrara, 2022-03-23:
and is it something that can already be done? to open and resize the card, or do you have to work on it ?, if so I know a bit of C # I can help

svenhb, 2022-03-24:
I will add an extra form to display the pen-down path.
Idea up to now: Open via menu - machnine control. Right click menu to setup background and path color, scaling and offset.
In principle end-switches must be enabled, to work with "fixed machine area":
[image]

Vincenzoferrara, 2022-03-24:
if you find this code useful and to open the form on the second screen in full screen

            if (Screen.AllScreens.Length > 1)
            {
                form.StartPosition = FormStartPosition.Manual;
                form.Location = Screen.AllScreens[1].WorkingArea.Location;
                form.FormBorderStyle = FormBorderStyle.None;
                form.WindowState = FormWindowState.Maximized;
                form.Show();
            }
            else form.Dispose();

searching online I found this formula for calculating dpi

// You need two things to calculate the physical DPI: the monitor size and the screen resolution.
            // For example, my monitor is about 18.5 inches wide and the horizontal resolution of the screen is 1680 pixels.
            // So my monitor's physical DPI is 1680px / 18.5in = 91dpi.

svenhb, 2022-03-25:
Ok, thanks, I will try.

Vincenzoferrara, 2022-04-02:
I notice a small bug, if I move the drawing, it is not updated on the projector until I click on the projector form.
and also that when I import a dxf the (main) form is resized.
I was also wondering if it was possible to have an automatic scale, where you set the measure of the dxf, and the measure of the projector.
a thousand thanks for all the commitment you put into this beautiful project!

svenhb, 2022-04-04:
Thanks for the feedback.
Could you explain a bit more the last two points? I don't get it...

>you can reopen the "background color" without closing and reopening
open the "background color" in the one screen.

svenhb, 2022-04-04:
>open the "background color" in the one screen.

I could move the whole projector setup to the main setup-form into a new tab...

Vincenzoferrara, 2022-04-04:
I would have other small requests if it were possible
  - import more than one drawing (in my case I would like to insert 2 dxf)
- copy button (I only found copy in clipboard but overwrite the current drawing)
- this is a question instead, is it possible to turn off the layers? (I use the dxf created by autocad)

I still rejoice for the effort you have to integrate all the functions
I'm recommending this beautiful project to everyone

svenhb, 2022-04-08:
I tried to implement your suggestions: https://github.com/svenhb/GRBL-Plotter/releases

Vincenzoferrara, 2022-04-11:
thanks it works perfectly.

besides a small suggestion,
-checkbox to disable modifying scales and examples
ceck is true = enable
so as not to change the measurement by mistake.

-if it is full screen, remove the windows bar (I think it's called that, so to speak the low one where you see the programs)

ps: instead to insert more files together is it possible?

Vincenzoferrara, 2022-04-13:
sorry with examples I mean the code below, so I write:

check box = click
scale.enable = false
 (to disable involuntary scale modification)

but for the dxf I had not noticed the check below.

there is also something to turn off the layers of the dxf?

even such a thing (see video) without having to reload all over again 

https://user-images.githubusercontent.com/64591523/163278517-c8ad3029-8988-418d-ab56-56db77d2f008.mp4

svenhb, 2022-04-16:
Can you provide a picture of your machine with the projector in action?

Vincenzoferrara, 2022-04-17:
currently the machine I am still assembling it (maybe I should have specified that it is not urgent)

svenhb, 2022-04-20:
Ok too bad, hopefully you make a picture when your machine is ready.
Check latest release for duplication option: https://github.com/svenhb/GRBL-Plotter/releases
[image]

Vincenzoferrara, 2022-04-24:
perfect!!!!!
if it were possible I would like a further implementation.
 the duplicate selcetion patch is fine, but if you want to have 6 you have to redo the command every time.

another thing missing (at least I think) are the layers, you can't "turn off" (I saw a video on youtube where I think there was this option.

BUG
- every time I open a drawing the form exits from full screen, and when you put it back everything becomes smaller (even the characters) os: windows 11

other functions that I don't need but would be nice to have in the future:

- If when you duplicate the drawing the distance between the 2 objects must never be less than the size of the cutter (or thickness of the knife / pen).

- Since the sensor is integrated, create a page for the calibration of the periodic plan (once a week / month / year)
and the acquired data are saved and automatically updated the gcode

svenhb, 2022-04-25:
> but if you want to have 6 you have to redo the command every time.

Can't you customize your original drawing? Isn't a graphics program easier to use?

>another thing missing (at least I think) are the layers, you can't "turn off" 

You can group the figures by layers, instead by colors, then delete the complete group with the layer you don't want.

>Since the sensor is integrated, create a page for the calibration of the periodic plan (once a week / month / year)
and the acquired data are saved and automatically updated the gcode

What kind of sensor do you mean? What calibtarion must be done?

Vincenzoferrara, 2022-04-28:
with sensor meant that of leveling 3d

Instead for multi copy (I don't know what to call it) I don't always need 6 pieces, it could be 6, 10, 20, etc so it would be more convenient to have a built-in function, but if it is too complicated for you I can also modify the design (I need it for work, so I must have the opportunity to be able to do many)

svenhb, 2022-04-28:
Next version will have offset for X and Y + optional selection width and height.
After a duplication (new with short cut "ctrl-D") the new inserted object is highlighted and selected.
So you can selected one object and duplicate as often you press ctrl-D - I think this is comfortable.

An other already existing option for the whole graphic is:
[image]

Vincenzoferrara, 2022-05-02:
ok thanks a lot, but can't you add a kind of counter?

I'll explain my situation a little better, in a file I have more or less 3-4 blocks (which will all have to be cut) let's pretend it's a puzzle that has only 4 pieces (even if it isn't) Now let's make fingers that for each of these 4 pieces I need 40, since and a block can you see how many there are in the drawing?  (Like com The right key that tells you how many copies there are) but perhaps it is not as easy as it seems
 PS: if you don't know how the AutoCAD blocks work I advise you to read them, because they have a very particular operation, which could make the implementation easier (always if you have time to do it)

svenhb, 2022-05-28:
The ID of the duplicated figure is increasing, thats all I have...

closed 2022-05-28 by svenhb