svenhb/GRBL-Plotter #264: CNC Com disconnection

YBougeard, 2022-03-21:
**Describe the bug**
Hello Sven, I'm Antoine Moineau's replacement and as such I'm continuing his work on the CNC machine he told you about.
We met a problem with the 1.6.5.2 release, the CNC Com often disconnects while the software streams the G-Code (it seems a bit random ?). The problem seems not to be specific to a particular DXF file.
Antoine mentionned this problem in another issue #255, the error in the logs is this one [logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/8314148/logfile.txt) : 

`2022-03-21 08:36:18.6229 | Error | GrblPlotter.ControlSerialForm  | SerialPortDataSend, Ethernet IO.IOException... System.IO.IOException: Le processus ne peut pas accéder au fichier 'G:\Affaires\1054-SLC\source\grbl-plotters\1.6.5.2\logfiles\logSendBuffer.nc', car il est en cours d'utilisation par un autre processus.`

Which means that the process couldn't access the file "logSendBuffer.nc" because it was already in use in another process, although there was no other apparent process running.
The problem doesn't exist with the 1.6.5.0 release, which we are currently using in the meantime.

**To Reproduce**
1. Pick a DXF file from [dxfFiles.zip](https://github.com/svenhb/GRBL-Plotter/files/8314154/dxfFiles.zip) and load it in GRBL-plotter 1.6.5.2 
2. Start streaming the G-Code
3. At some point, the CNC Com will disconnect, if not, you can try to restream the G-Code to reproduce the problem

**Expected behavior**
The CNC Com shouldn't disconnect during the G-code streaming

**Screenshots**

[image]

Please tell me if you need more information or anything else !
Thank you for your help !

YBougeard, 2022-03-21:
Alright thank you, I will try it out !

svenhb, 2022-04-01:
Should be solved: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.6.0

closed 2022-04-08 by svenhb