svenhb/GRBL-Plotter #265: Camera - Tool coordinate teaching goes away if you set work coord 0:0 for fiducial

yagizzha, 2022-03-21:
https://grbl-plotter.de/data/uploads/pic/plotter_fiducials/fiducial_xy_auto.gif
when following this , if you do the "set work coordinate 0;0 here , it removes the teachings from this window:
[image]

svenhb, 2022-03-21:
I never tried with "Camera mounted on Z axis" and assume that the offset is applied to the wrong coordinate system (probably the currently active one)...
I need to figure out...

yagizzha, 2022-03-21:
[image]
When swapped to tool coordinate with move stage , goes where camera was , as we expect , making me think the calibration is succesful.
[image]

[image]
But then if we do the "set work coordinate 0;0 here" it ruins all the previous steps.

svenhb, 2022-03-21:
The step "set work coordinate 0;0 here" will be done first, to be close to the fudcial - as a rough teaching.
Then the "automatic fiducial correction" will be started.

You are right, if you do then again a "set work coordinate 0;0 here" you loose the previous teaching - so don't do it :-)

yagizzha, 2022-03-21:
if we do the "set work coordinate 0;0 here" first , how will the process go ? 
wont it be "set0;0-->calibrate-->cut", in that case we'd have to calibrate with every cut since we likely wont be putting it in an exact position . Is there any way to set up tool distance manually ? or have it as a "relative" system rather then a fixed coordinate system ?

closed 2022-03-29 by yagizzha