svenhb/GRBL-Plotter #266: Circle detection doesnt work on zip version

yagizzha, 2022-03-22:
[image]
Circle detection isnt working when using the zip version , using .net 4 , VS2019 , works fine with the exe version

svenhb, 2022-03-23:
That's strange, the Zip and the installer are using the same files. Most probably the settings are not identical, when starting from different locations. 
You may check C:\user\xyz\AppData\Local\GRBL-Plotter\GRBL-Plotter.exe_Url_....\Version\user.config... for the date of stored cofigurations, to check if different files are used, when starting from different locations...

[image]

yagizzha, 2022-03-23:
Weirdly , it keeps resetting my values to this as soon as i open the camera form :
[image]
Thought maybe it was because of something i changed in the code so downloaded a fresh version but same problem still occurs.

yagizzha, 2022-03-23:
It changes when you click "paper" for filters , which i guess is expected but still doesnt detect

svenhb, 2022-03-23:
Can you compare files between installed (see logfile for location) and zipped version? If they are identical, it must work.
I tried at work with simulation picture and it works from both location (installed and copied):
[image]

Perhaps the teaching for Z mounted camera is missing when starting the zip-version?

yagizzha, 2022-03-23:
Made 
[image]
max side bigger and its working now , however i have no clue why zip and installed reacts diffrently since they had the same parameters.
[image]
also there is a false detection , also on VS2019 clicking "save" or "load" next to filter settings was crashing the problem , works fine on VS2022

svenhb, 2022-03-23:
Probably the scaling between both are different: do you get the same circle diameter in the picture, if set to e.g. 10?
[image]

yagizzha, 2022-03-23:
yeah i do , i've set both to 10 then calibrated it , had a very weird bug tho , randomly fiducial circles on svg got bigger , calibrated again and it fixed itself after

yagizzha, 2022-03-29:
Since problem itself is solved , im closing this

closed 2022-03-29 by yagizzha