svenhb/GRBL-Plotter #267: Maximum feedrate limited

diemanvanstaal, 2022-03-24:
The maximum feed rate seems to be limited to 2000 mm/min irrespective of the limit set in the setup menu.
In my case I would like to set the feedrate to 6000 mm/min. 
The firmware is set for 6000 mm/min.
The XY Feedrate under the setup\ G-Code Generation menu is set to 6000 (not from Tooltable).
When I stream the g-code, the feedrate is limited to 2000 mm/min. The feed override allows me to increase the feedrate by 100% to 4000 mm/min but no more than that.
Is there a setting I am not aware of that will allow faster feedrates?

GRBL Plotter V1.6.5.2
GRBL V1.1h

Thanks

svenhb, 2022-03-25:
If the GCode shows F6000, the reason for the limitation is inside the grbl-settings, please check your max. feedrates $110, $111, as described here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings

Just click the "$$" button in the bottom of the COM-CNC window to get the list.

diemanvanstaal, 2022-03-25:
Thank you Sven,

I had the G-Code header set to F2000.

I thought the Feedrate was set by the XY Feedrate entry.

 

Now that I changed the header to F6000 I get the expected rates from the machine.

 

Thank you for your support and excellent software.

-Christo

svenhb, 2022-03-25:
Thank you,
but strange: actually the XY-Feedrate (which is applied when switching form G0 to G1 command) should be applied and used by grbl...

[image]

The value in GCode header is more for safety, if you send a G1 command without a F-value grbl gives an error, if F wasn't set before.

diemanvanstaal, 2022-03-25:
Hi Sven,

Thanks for your assistance.
I am new to all of this, so I’m not sure why it behaves like this.
Here is the G-Code and setup file (with F6000 header) and firmware settings if you are interested:

**** Connected to COM8 @ 115200 baud ****
Grbl 1.1h ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 2    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 320.400    (X-axis travel resolution, step/mm)
$101 = 318.400    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 6000.000    (X-axis maximum rate, mm/min)
$111 = 6000.000    (Y-axis maximum rate, mm/min)
$112 = 2000.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 500.000    (X-axis maximum travel, millimeters)
$131 = 500.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
Ok

Regards,
Christo
[Setup.zip](https://github.com/svenhb/GRBL-Plotter/files/8350637/Setup.zip)

svenhb, 2022-03-25:
Thanks for the files, nothing unusal to find in it. 
Need to check my code, why the F command is missing in the first line of G01...
Your GCode should look like
```
(<Figure Id="1" Geometry="m" PenColor="black" PenWidth="0.265" PathID="path10" PathLength="40137.1"> )
G00 X95.615 Y95.615
M08
G01 X176.569 Y50.742 F6000

```

svenhb, 2022-04-20:
No further feedback, no problem anymore