svenhb/GRBL-Plotter #268: G-Code 4 axis convert to Marlin V.2.09

qulet, 2022-03-27:
Please let me know how to adjust the G-code for the A-axis tangent knife so that it can be made from Marlin v2.09. (Just changing the letter of the A axis to E is not enough). Is it possible to add a connection with Marlin to the program in order to use only the CNC 4 axis functions, omitting the heating function?

svenhb, 2022-03-28:
How needs the code to look like?

qulet, 2022-03-28:
G-code for the 4th axis of the tangential knife after a slight modification {ASCI "conversion "(" to ";" and "A" to "E") allows you to run the code in the Marlin software. My request is to add a connection to the Marlin firmware as it is done in program "LaserGRBL" Your GRBL_Plotter program is very good, adding this function would make it even BETTER ...

svenhb, 2022-03-29:
>My request is to add a connection to the Marlin firmware

I see. I added this feature in 1.5.4.0 but got no feedback.... https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.5.4.0

qulet, 2022-03-30:
234 / 5 000
Wyniki tłumaczenia
I am currently using GRBL-PLOTTER 1.6.5.2 and despite the change:
I can't connect to the Marlin 2.09.01 firmware from your software. From the LaserGRBL programs to the switch to the Marlin software, this connection works. 
[image]

svenhb, 2022-04-01:
Perhaps no it works: I switched line-end marker from \r to \n. I have no hardware to check function.

qulet, 2022-04-03:
[image]
Niestety nie udało się nawiązać połączenia z Marlinem :(((

svenhb, 2022-04-04:
Are you sure it is the right baud rate? I read about 250000 instead of 115200. 
I never used LaserGRBL, so I don't know what the message means: Sounds to me it is also not connected...
[image]

qulet, 2022-04-04:
In LaserGRBL, the correct Firmware must be selected: GRBL; Marlin or Smoothie.
I chose Marlin with 115200 baud rate. Pressing the button - connect highlights the jog icons in green and it is possible to control the motors of the X, Y, Z axes.

svenhb, 2022-04-08:
I could load Marlin on my Mega 2560 and improove behavior of GRBL-Plotter, please check:
https://github.com/svenhb/GRBL-Plotter/releases

qulet, 2022-04-09:
The connection worked !!! on these settings:
[image]

qulet, 2022-04-10:
My joy was a little before early ...
I have two printers:
1. Marlin 1.9 Ramps1.3 firmware. The connection is working
[image]

2. Firmware Marlin 2.09 Mks-Sbase1.3. The connection is not working.
[image]
I want to connect with the Marlin 2.0 MKS-Sbase1.3 32bit firmware. This printer has the ability to manually change the extruder to any other tool: LED laser, tangential knife. Unfortunately, here the connection does not work.

svenhb, 2022-04-11:
Ok, too bad...
Are you sure the 2nd printer uses same baud rate?
Did you try disconnect/connect button?
Did you try restart of GRBL-Plotter?

I assume yes...

In the start up I miss the "unknown command" as for the 1st printer
Perhaps you can figure out, what are the first messages? Perhaps finally I need to add a checkbox to force marlin and not try to automatic recognize
[image]

qulet, 2022-04-12:
In Marlin 2.0 the transfer was changed to 25,000 to be the same as in Marlin 1.9. Of course, I tried to change this value and restart GRBLPlotter, but unfortunately it did not allow the connection to be established. On ramps 1.3, after activating RST, values such as the M501 g-code were loaded. Unfortunately, it was not possible to establish such a connection on the MKS SBase.

svenhb, 2022-04-13:
Check new release 1.6.6.2: https://github.com/svenhb/GRBL-Plotter/releases
[image]

qulet, 2022-04-14:
Connection to Marlin 2.0 interface 4 axis Control / Jogging works after connection is established. G-Code Streaming does not work ...
[image]

svenhb, 2022-04-15:
you may switch on this options to get more informations
[image]

qulet, 2022-04-15:
[image]
[image]
[image]

svenhb, 2022-04-20:
I made another change, perhaps no it works?: https://github.com/svenhb/GRBL-Plotter/releases

qulet, 2022-04-20:
[image]
[image]
[lastProcessed_.zip](https://github.com/svenhb/GRBL-Plotter/files/8523918/lastProcessed_.zip)
SVEN sorry to tire you so much with these amendments. Maybe I'm doing something wrong?
The latest version connects to Marlin just like the previous one. Unfortunately, G-code streaming does not work.

svenhb, 2022-04-21:
Did you try to press "RST"?
[image]
You also change the streaming method:
[image]

qulet, 2022-04-21:
All the functions of your program work fine on Marlin 1.9 Ramps:
[image]
![MarlinRamps_0](https://user-images.githubusercontent.com/30979927/164510937-e528b77c-b5b4-425c-89e4-
a26e0bf7a9d5.jpg)

The problem is Marlin 2.0 MKS-Sbase. There you can get a connection sometimes, jogging works, but there is a blank field in the connection status ... and g-code forwarding is not working.There, for the connection to work (jogging), I have to press RESET in the program window. But the connection status is blank and g-code uploading is not working.
[image]

svenhb, 2022-04-22:
>The problem is Marlin 2.0 MKS-Sbase.

Strange, there is no start-up message after hard-reset.
Do you have a HEX-file of this firmware, which I can upload to my Arduino Mega-2560?
As shown above I also use Marlin 2.0

qulet, 2022-04-22:
I use Visual Studio Code to compile Marlin for MKS-Sbase CD. The output file there is the file: firmware. bin. There must be a different pinout for ramps. I never compiled Marlin 2.0 for ramps to get the Hex peak. Enclosed my Marlin 2 configuration for the MKS-Sbase board.
[Uploading M_2.0 MKS-SBASE_config.zip…]()

qulet, 2022-04-23:
As soon as RST is pressed in the main program window, jogging starts, connection status clean button. how will i do a RESET on the board: the message [ERROR] Closing port pops up
[image]

qulet, 2022-04-23:
Svenhb with compiled GRBL LPC Ceprezzi. I have uploaded the firmware.bin file to the MKS-Sbase SD card.
Everything about GRBLPlotter on this firmware seems to be working fine. I have to accept that if I want to use your software, I have to change the Marlin firmware to the GRBL LPC firmware. Thank you for your help.

svenhb, 2022-04-23:
I'm happy to hear this. Marlin is really strange and I have no idea what's going on with this MKS board...

closed 2022-04-23 by svenhb