svenhb/GRBL-Plotter #269: Adjusting offset between tiles with a camera

YBougeard, 2022-03-29:
Hi, I'm using GRBL-Plotter to import and process rather long DXF files to cut long pieces of fabric and I'm currently having troubles with connecting shapes between tiles.
Most of the time from one tile to another there's a 1-2 mm offset (sometimes more) on the Y axis due to our mechanical setup. I would like to use a camera and potentially a homemade shape recognition software to correct this offset and succesfuly connect forms between tiles.
My idea to solve this is to have the software check a mark the camera at the end of each tile and apply a Y offset on the next tile if necessary, do you think it would be possible ? 

GRBL-Plotter already has a shape recognition feature, maybe it could be adapted to solve my problem ?
Having the possibility to draw small circles or a few lines at the end of the tiles when needed to connect shapes might do the trick, but I would need to test that on small drawn shapes as well.

Here are some example DXF files I would like to process but the straight lines end up being discontinuous.
[dxfFiles.zip](https://github.com/svenhb/GRBL-Plotter/files/8369014/dxfFiles.zip)

If you have any questions, please do not hesitate to ask.
Thank you for considering my suggestions.

svenhb, 2022-03-29:
Hi,
Isn't your workflow like: 

- processing one tile, 
- Send command via 3rd serial to move Y position
- Wait for ready from 3rd serial until position reached
- processing next tile?

I would better think of correcting the mechanical Y position externally, before sending the "ready" signal via the 3rd serial COM.

YBougeard, 2022-03-30:
Hi,
Thank you for your quick response,
Yes, this is precisely my workflow, however mechanicaly adjusting the machine or fabric is tedious (and inaccurate) considering our setup and also considering the potential size of our dxf files (sometimes more than 70 tiles)...
That's why I thought that adjusting the offset from the software would be a more ergonomic way of dealing with the issue.

svenhb, 2022-04-20:
As already written, you need to improove your mechanics.

closed 2022-04-20 by svenhb