svenhb/GRBL-Plotter #270: Circles in cutting mode are bugged

yagizzha, 2022-03-29:
Using svg 
[image]
this circle starts out line this , which is what we expect but it gradually gets worse and becomes like this :
[image]

SVG used was this : 
[image]
Also even tho 2 circles were marked as fid1 and fid3 in id , they werent detected , i can open a new issues for that if needed ?

svenhb, 2022-03-29:
I don't understand the first part of your issue...

2nd part: not the id but the layer must be named - setup:
[image]

My example "graphic_fiducial_circle.svg":
[image]

yagizzha, 2022-03-29:
First part of issue is the cutting part goes in angles that wont cut the paper but tear it instead like it is at the end , it starts out at a correct angle but slowly gets worse and at the end its 90 degrees of what it should be

svenhb, 2022-03-29:
It is just the simulation, I need to check. If you avoid G2/G3 it looks correct:
[image]

svenhb, 2022-03-29:
Ok it is a bug, when using G2/G3: it starts at A90 and should end at A(90+360) = 450, but ends at 360...

svenhb, 2022-05-28:
No feedback? Seems to work now...

closed 2022-05-28 by svenhb