svenhb/GRBL-Plotter #271: While cutting circles , A axis starts to move weirdly sometimes

yagizzha, 2022-03-30:
During this cut (its supposed to be a circle , because it was for testing calibrations werent done perfectly) it does a weird clicky-ish sound while cutting , not going smoothly 
https://user-images.githubusercontent.com/88779645/160825346-6bc63b26-6850-455f-a2a6-b6f6b6166ec8.mp4
Gcode for that one looks like this :
[image]
However , another circle from same file works fine (again dont mind the paper moving , its just to show the movement):
https://user-images.githubusercontent.com/88779645/160825636-6576d7fd-1b55-4dbb-b293-2c764c4f0477.mp4
GCode for that one is 
[image]
 Is this a bug or some feature we are missing ? if a feature how to work around so it cuts smoothly aswell

svenhb, 2022-03-30:
You need to increase this value:
If angle between two lines is too big (bigger than this value), the start angle for the second line will be adapted.
[image]

yagizzha, 2022-04-23:
Same issue happened while doing it with ellipses , increasing start angle didnt help

svenhb, 2022-04-23:
Please send graphic, gcode.result and expected behavior

yagizzha, 2022-04-25:
GCode: ( Use case: ? )
( SVG import by GRBL-Plotter 1.6.5.2 )
( Source: C:\Users\Owner\Desktop\tests\F23.svg )
(<Header >)
( G-Code lines: 222 )
( Pen Down/Up : 15 times )
( Duration ca.: 1.0 min. )
( Conv. time  : 00:00:00.1230846 )
( Original graphic dimension min:1.049;1.052  max:303.769;155.675)
(</Header >)
(<Tangential Axis="A"/>)
G54 (Setup - GCode-Header)
G90
G00 Z3.000
M3 S1000 
(<Figure Id="1" Geometry="circle" Layer="fiducial" PathLength="30.7"> )
G00 X9.793 Y4.896 A4.262  
G01 Z-1.111 F1000 (PD)
G01 X9.690 Y5.896 A4.262 F2000 
G01 X9.385 Y6.854 A4.786  
G01 X8.890 Y7.729 A5.309  
G01 X8.228 Y8.485 A5.833  
G01 X7.425 Y9.090 A6.357  
G01 X6.515 Y9.517 A6.880  
G01 X5.537 Y9.751 A7.404  
G01 X4.532 Y9.779 A7.928  
G01 X3.542 Y9.602 A8.452  
G01 X2.610 Y9.226 A8.975  
G01 X1.774 Y8.668 A9.499  
G01 X1.069 Y7.950 A10.023  
G01 X0.526 Y7.104 A10.547  
G01 X0.167 Y6.165 A11.070  
G01 X0.008 Y5.173 A11.594  
G01 X0.054 Y4.169 A12.118  
G01 X0.305 Y3.195 A12.642  
G01 X0.749 Y2.293 A13.165  
G01 X1.368 Y1.501 A13.689  
G01 X2.136 Y0.852 A14.213  
G01 X3.020 Y0.373 A14.736  
G01 X3.983 Y0.085 A15.260  
G01 X4.985 Y0.000 A15.784  
G01 X5.983 Y0.121 A16.308  
G01 X6.935 Y0.444 A16.831  
G01 X7.801 Y0.954 A17.355  
G01 X8.545 Y1.630 A17.879  
G01 X9.135 Y2.444 A18.403  
G01 X9.547 Y3.361 A18.926  
G01 X9.762 Y4.343 A19.450  
G01 X9.793 Y4.896 A19.856  
G00 Z3.000 (PU)
(</Figure>)
(<Figure Id="2" Geometry="polygon" Layer="Katman_x0020_1" PathLength="31.3"> )
G00 X88.490 Y67.522 A-1.347  
G01 Z-1.111 F1000 (PD)
G01 X101.956 Y59.650 A-1.347 F2000 
G01 X115.596 Y51.778 A-1.333  
G00 Z3.000 (PU)
G00 X115.596 Y51.778 A-4.000  
G01 Z-1.111 F1000 (PD)
G01 X115.596 Y20.465 A-4.000 F2000 
G00 Z3.000 (PU)
G00 X115.596 Y20.465 A-6.692  
G01 Z-1.111 F1000 (PD)
G01 X101.956 Y12.768 A-6.692 F2000 
G01 X88.490 Y4.896 A-6.653  
G00 Z3.000 (PU)
G00 X88.490 Y4.896 A-9.333  
G01 Z-1.111 F1000 (PD)
G01 X74.849 Y12.768 A-9.333 F2000 
G01 X61.208 Y20.465 A-9.308  
[... 158 lines cut ...]
G01 X294.119 Y146.346 A13.689  
G01 X294.887 Y145.696 A14.213  
G01 X295.771 Y145.218 A14.736  
G01 X296.734 Y144.930 A15.260  
G01 X297.736 Y144.845 A15.784  
G01 X298.734 Y144.966 A16.308  
G01 X299.686 Y145.288 A16.831  
G01 X300.552 Y145.798 A17.355  
G01 X301.296 Y146.474 A17.879  
G01 X301.886 Y147.288 A18.403  
G01 X302.298 Y148.206 A18.926  
G01 X302.513 Y149.188 A19.450  
G01 X302.544 Y149.740 A19.856  
G00 Z3.000 (PU)
(</Figure>)
M05
M30
File : 
[image]

This is the one where we have problem :
[image]

This is how the cut ends up being , expected was it being smooth:
[image]

svenhb, 2022-04-25:
If I copy & paste your GCode I see the same edges in the 2D-view, as in your picture:
[image]

You should adapt this parameters:
[image]

Then curve is less angular:
[image]

svenhb, 2022-05-28:
No further feedback, seems to work now...

closed 2022-05-28 by svenhb