svenhb/GRBL-Plotter #272: When using camera form to correct fiducials , cutting piece angle gets bugged

yagizzha, 2022-03-30:
When cutting without any camera input it follows like this :
[image]
But after fiducial correction it becomes : 
[image]

svenhb, 2022-03-31:
I never tried fiducial correction together with tangential axis. 
I assume, you are not using 360° for a full turn, so the applied rotation-offset to the tangential axis would be wrong...
[image]

svenhb, 2022-03-31:
Also you need to click into the 2D view to set the marker again

yagizzha, 2022-03-31:
Units per turn on our machine is set to 16 since that causes a 360 degree turn . 

The problem is the angle of the knife starts wrong and goes like that all the way through the cutting process when using fiducial detection .

yagizzha, 2022-04-01:
If its a bug spesific to my case , can you give me the line where knife angle gets calculated so i can try to fix ?

svenhb, 2022-04-01:
I already fixed it, I plan to release the new version this evening or tomorrow - I still working on another part of the software.

If you can't wait:
In /MachineControl/GCodeTransform.cs - line 269, change the block to this:
```
                    if (tangentialAxisEnable)
                    {
			double tangle =((double)Properties.Settings.Default.importGCTangentialTurn * angle / 360);

                        if ((tangentialAxisName == "C") && (gcline.c != null)) { gcline.c += tangle; }
                        else if ((tangentialAxisName == "B") && (gcline.b != null)) { gcline.b += tangle; }
                        else if ((tangentialAxisName == "A") && (gcline.a != null)) { gcline.a += tangle; }
                        else if ((tangentialAxisName == "Z") && (gcline.z != null)) { gcline.z += tangle; }
                    }

```

yagizzha, 2022-04-01:
that worked perfectly, thanks

closed 2022-04-01 by yagizzha