svenhb/GRBL-Plotter #274: Move the graphics path accurately

lianzaozi, 2022-04-04:
As shown below, "Apply offset to GCode"not available.By selecting points on the screen it is sometimes not possible to move the graph exactly where you want it.

I have an idea to quickly and accurately move the graph to the desired position. First, select the graph to be moved, and then right-click to add a window in the pop-up window to input the target value (X coordinate value and Y coordinate value) to which the graph should be moved. , click OK to move.

[image]

lianzaozi, 2022-04-04:
I know, select the graphic, hold down the middle mouse button to move,“Apply offset to GCode” activate available.If you can add an input X coordinate, Y coordinate to move the selected graphics, I think it will be more efficient

svenhb, 2022-04-04:
I think you use an old version. Since 1.6.5.0 the "Apply offset" entry was removed. Also you will see handles at the selected figure, to drag, rotate or scale the selection:
[image]

lianzaozi, 2022-04-06:
For how to move and edit graphics in the field of laser cutting, and laser process changes, you can refer to the software design in the "file" folder, or future versions of the plotter software may add some similar functions that are easy to use in this software.

For details, please see the contents of the compressed package below

[file1.zip](https://github.com/svenhb/GRBL-Plotter/files/8423071/file1.zip)

svenhb, 2022-04-20:
Check new item propertie of selected path: https://github.com/svenhb/GRBL-Plotter/releases
Finally I don't want to make a graphic editor, better use a real vector graphic program - or the program you suggested.
[image]

lianzaozi, 2022-04-20:
It does take a lot of work to develop drawing software, and there are already some source code available so we don't have to start from scratch. You can use the source code for "SimpleCAD", which is developed in C#. If you can't find the source code, you can send me your email address and I'll send you the source code

lianzaozi, 2022-04-20:
SimpleCAD software can read and write DXF files with "NetDXF".

svenhb, 2022-05-28:
I don't want to make a drawing software, sorry...

closed 2022-05-28 by svenhb