svenhb/GRBL-Plotter #275: Detection inside svg files taking id aswell

yagizzha, 2022-04-04:
Program we are using right now doesnt add labels to our svg files so we cant use the output directly without changing them . Would it be possible to mark fiducial layer with id ? or any other solution to this . If there is a line in code that we can look at for this that'd help too .

svenhb, 2022-04-04:
Some more changes are needed, right now I can say - without any tests:
Add "PathId" in MachineControl\GCodeAnalysis.cs line 751:
[image]

In GCodeCreation\XmlMarker.cs change variable "PathId" from int to string. 
In function SetBlockData change accoring new type.
tmp.PathId = "";
if (element.Contains("PathId")) { tmp.PathId = GetAttributeValue(element, "PathId"); }

Perhaps thats all

svenhb, 2022-04-05:
There is a small bug: the created XML-Tag is "PathI**D**" but, as shown above, variable is looking for "PathI**d**"
Please change line 547 in GCodeCreation\Graphic2GCode.cs to
`            if (pathObject.Info.PathId.Length > 0) attributes.Append(string.Format(" PathId=\"{0}\"", pathObject.Info.PathId));
`

yagizzha, 2022-04-06:
Thanks that worked perfectly !

closed 2022-04-06 by yagizzha