svenhb/GRBL-Plotter #281: Windows 64 bit updated.

78Matteo78, 2022-04-19:
hello svhen I wanted to ask you if it is possible to update grbl plotter to the 64-bit platform as well. I tested on Windows 10 and it struggles to run. then as regards an old problem of speed of the axes, which above 600 went half as if it were 300, since I use a mega 2560 clone, I solved it with the firmware update you published. Thanks for everything

svenhb, 2022-04-19:
What is the advantage of 64-bit version? I had no problems up to now with 64-bit Win-10 operating systems.

78Matteo78, 2022-04-19:
I am using an old version, 1.3.3.4 because it has no problems in the sense that I can open large gcode files But the other versions, starting from the 1.4-1.5-1.6 series, fail to load the gcode and slow down all the program functions. the program is also installed in the x86 folder for compatibility but new versions don't work. I wanted to let you know that with the new firmware I have also solved communication problem, often with error 25 in the COM window. Greetings Matteo

svenhb, 2022-04-19:
Bad to hear you have problems to open large gcode files - are they "pure" gcode, or with my XML-Tag extensions?
Can you upload or email me such a large file, to find the problem? Perhaps I can add an option to bypass parsing uneeded features...
Good to hear that other firmware could fix a problem.

78Matteo78, 2022-04-19:
With versione 1.3.3.4 in 10 seconds open file martedì,

78Matteo78, 2022-04-19:
ok svhen, software 1.6.6.2 works fine even in 64 bits. All functions work well. There remains the problem of the file being too large and not opening. Please delete from the list of problems .... Thanks greetings Matteo

svenhb, 2022-04-19:
>software 1.6.6.2 works fine even in 64 bits

Good to hear,

>There remains the problem of the file being too large and not opening

Can you upload or email me such a large file?

78Matteo78, 2022-04-19:
I sent you 2 emails with the file attached, did they arrive?

svenhb, 2022-04-20:
no, I got no email

78Matteo78, 2022-04-20:
Big file gcode
--
Inviato da Libero Mail per Android mercoledì,

78Matteo78, 2022-04-20:
perhaps files with the extension. nc are not accepted ??

78Matteo78, 2022-04-20:
however it is an image file converted into gcode with grbl laser with 3 lines per millimeter. if you don't receive the file I convert it to txt and try to send it to you

78Matteo78, 2022-04-20:
or I try to zip

78Matteo78, 2022-04-20:
> can you "zip" it? It will reduce the size.

Zip file. 
--
Inviato da Libero Mail per Android mercoledì,

svenhb, 2022-04-20:
no... right email adress? svenhb@web.de

78Matteo78, 2022-04-20:
Now?

svenhb, 2022-04-20:
yes, now I got it - thanks.
1st try: after 1 minute still not displayed :-(

78Matteo78, 2022-04-20:
Tanks svenh!!!

svenhb, 2022-04-20:
Ok, to bad. Your example has 91500 lines, at 100000 lines GRBL-Plotter skips time consuming display features....
I will make this limit changeable in the setup....
Time consuming features are:

- direction arrow and id at pen-up path
- try to colorize pen-down path by tool-table settings

svenhb, 2022-04-20:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases
[image]

svenhb, 2022-04-20:
Thank you, 
I hope the loading time is now ok?

78Matteo78, 2022-04-20:
Yes yes.... Perfect

svenhb, 2022-05-28:
Seems to work...

closed 2022-05-28 by svenhb