svenhb/GRBL-Plotter #282: Wire bending mode

Ams-Laser, 2022-04-22:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
is there a way to have feed and bend support for a wire bending machine
there is already a tangential knife support, so angles to "bend" are already calculated, it is just a matter of adding springback compensation and handle the bending peg

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

svenhb, 2022-04-22:
I was also thinking about such a wire-bender - my problem: how to create content in 2D view (SVG, DXF) which I can export to wire-bender data. Free dimensions to calculate turn-angle are line-color and line-width.
Your machine only bend in 2D dimension, without turning the wire?

>there is already a tangential knife support, so angles to "bend" are already calculated, it is just a matter of adding springback compensation and handle the bending peg

No problem. Is it a fix add-on in angle or depending on the bent-angle = a percentage value?

Ams-Laser, 2022-04-22:
springback varies with material, best guess would be to be able to use both
mabe a checkmark use springback compensation and value then another two check marks one indicating value is percent other indicating is angle in degrees

the peg that does the bending also needs to account for the offsetradius of the peg, and a gcode block to duck (lower) the peg to go under the wire when bends ara the other way

Ams-Laser, 2022-04-22:
look at this https://youtu.be/SQIYZ-iKuG4

svenhb, 2022-04-22:
Yes, nice.
To make letters out of a stripe of material, by making grooves at the bending positions, I made an add-on. Check this "issue": https://github.com/svenhb/GRBL-Plotter/issues/220

Ams-Laser, 2022-04-22:
would it be possible to conbine that with the actual bending of the profile ?

Ams-Laser, 2022-04-22:
the pegs could be either solenoid or axis control
it is better for the diferent setups if we leave a "pull pegs code" "push pegs out code" input section (for custom gcode to retract or extend the pegs, also a single or double peg tik march is needed
same for the cut wire command "cut wire code" as i mighy need to feed in more material to get to the cut location then either axis or solenoid used to do the cut

svenhb, 2022-04-23:
Try this first version with fix axis X for feed and Y for bend angle...
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1664a.exe
[image]

Ams-Laser, 2022-04-23:
Hello,
Thanks!!!!!!!
on machine parameters, the bend radius is the peg radius ?
there is a grayed wire diameter setting ?

svenhb, 2022-04-23:
>on machine parameters, the bend radius is the peg radius ?

The radius of the bent wire: probably peg position from center of turning wheel, minus peg-radius

>there is a grayed wire diameter setting ?

Perhaps it is needed later on, for fine tuning the algorithm

svenhb, 2022-04-28:
Any results up to now. you could show?

Ams-Laser, 2022-04-30:
Hello, 
im finishing the hardware and will get back with feedback

Thanks
Ed

svenhb, 2022-05-28:
Still no feedback, seems to work...

closed 2022-05-28 by svenhb

ivofuture, 2023-03-28:
I'm going to make the Bending channel letters machine and could you tell me if I charge marlin or any version of grbl? 2 weeks and you're good to go.

svenhb, 2023-03-28:
If you want to use GRBL-Plotter, better go with grbl. Communication with marlin is not well tested and a bit strange in my eyes.
Any feedback is welcome.

ivofuture, 2023-03-28:
Excellent start to work.