svenhb/GRBL-Plotter #283: Pen Clamp Servo not working in tool change script but does via console

mortonhoby, 2022-04-24:
When using M3 S100 to release the pen clamp (or M3 S0 to clamp) in a script there command doesnt execute (or it doesnt have time to), but if I issue the same command via 'SEND' in the console it does correctly drive the servo.  Is there a way to add a delay or is there another setting I need?

svenhb, 2022-04-25:
If PWM is enabled as Pen-Up/down function, you can add a delay, which results to G-Code "G4 P0.5" for a 0.5 second delay.
Please show your scripts and GCode.

mortonhoby, 2022-04-25:
Hi Sven,

 

Many thanks for your response.   I have been following the implementation on a CNC Plotter with Auto Tool changer by Dejan and had used all of his inputs, however I had missed setting the spindle speed to ‘0’ (I had it at 1000) and this seemed to cure the problem.

 

I had been trying ‘G4 X1’ (Bosch style G code for 1 second delay) and ‘G4 F1’ (Fanuc style G code) I had not seen your suggested ‘G4 Pn’ method before, which prompts me to ask if there is a list of accepted G codes for the GRBL Plotter?

 

On another matter is it possible to position ‘text’ other than at 0,0 (or indeed move it)?

 

Thank you in advance, and thank you for developing a great open source project.

 

Regards,

 

Roger Morton

svenhb, 2022-04-25:
>if there is a list of accepted G codes for the GRBL Plotter?

Finally the grbl firmware on the controller needs to accept the commands, check this list: https://github.com/gnea/grbl/wiki

>On another matter is it possible to position ‘text’ other than at 0,0 (or indeed move it)?

You can click on the text and should get handles to move, scale or rotate the object.

mortonhoby, 2022-04-25:
Hi Sven,

 

Once again many thanks.   The links help a great deal.

 

Regards,

 

Roger Morton

svenhb, 2022-05-28:
Seems to be fixed...

closed 2022-05-28 by svenhb