svenhb/GRBL-Plotter #284: tangential axis and drag knife doesnot works simultaneously

huley76, 2022-05-05:
As suggested , i made enabled both tangential knife and drag knife options for my tangential knife operation. Still the g code that i create from "Create simple shapes" doesnot take account into for the offset and the g code and figure i get is unbroken therefore everytime the knife is raised and lowerd , a gap appears due to the offset

[image]

[image]

How i can activate the feature to compensate the offset and use tangential knife at the same time.

P.S. how can i choose the depth of cut and number of passes. I want to cut 2mm paper in one cut and now It cuts  with 1.1 depth and in two pass even if i set Z min as 2.
 Thank you

svenhb, 2022-05-05:
Here you need to put in the offset from center (tip of knife to axis of rotation): 
[image]
[image]

[image]

If you want to cut same path several times:
[image]

huley76, 2022-05-07:
I have adjusted the settings as you recommended but still it doesnot work for me.
[image]

The problem that i have is the while the knife is raised and rotated it lands not on the position where the cut was, but a little behind as shown below.
[image]

I looked at one of the issue with same problem as mine where you solved this problem recently and that was made available recently in new version, as you can see i installed the newest version  but i still couldnot get the same result which was somewhat like this
[image]
:

how can i activate this feature?

svenhb, 2022-05-07:
You need to set "Offset from center" very exactly

huley76, 2022-05-07:
Mine is almost same as you used in video for tangential knife

[image]

I even tried with different values but the offset remains the same which is my mechanical offset of knife holder

svenhb, 2022-05-07:
Looks like your knife-tip is in front of the rotation axis - sorry not supported.
Your tip must be BEHIND the rotation axis.
[image]

huley76, 2022-05-07:
Do you have an example of it?  Mine is almost in the same plane with the rotary axis.

[image]
"Your tip must be BEHIND the rotation axis." do you mean by this or how can i achieve this

svenhb, 2022-05-07:
Almost is not good enough.
Either the tip is on the rotation axis (center exemple) then you don't need to enable "drag tool".
Or right example, then "drag tool" must be enabled.
[image]

huley76, 2022-05-07:
Thank you for the help. I will impliment the modification necessary.

huley76, 2022-05-08:
Now i made changes as in right example above still the issue is same because the software doesnot accounts for offset. In the GUI the line is not broken therefore it rotates and the blades moves down and starts cutting, whereas in my case it should rotate go further a little to compensate the offset and only then the knife should be lowered
https://user-images.githubusercontent.com/105011612/167293842-8fbc688e-d021-473c-a6ca-bcb206bd0fa3.mp4

svenhb, 2022-05-08:
Ok, now I see an offset of ca. 8 mm in the correct direction, so I would try this settings:
[image]
[image]
The path should look like this, to take care of the offset:
[image]

huley76, 2022-05-08:
which version of the grbl are you using here? tried the identical setting in setup , but the path is not like yours but continuous.

svenhb, 2022-05-09:
Ok, sorry. At least it is implemented here: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1664a.exe

svenhb, 2022-05-11:
Any progress?

svenhb, 2022-05-28:
No feedback, problem seems to be solved...

closed 2022-05-28 by svenhb