svenhb/GRBL-Plotter #285: Sometimes dxf files execution stops (version 1.6.5.0)

YBougeard, 2022-05-06:
Hello,

I'm currently using version 1.6.5.0 which works fine most of the time.
I'm trying to identify the causes of an issue that happens every once in a while : We load a dxf file, launch it, and the execution stops at some point (most of the time, rather late in the streaming).
You can see in the logs (at the end of the linked file) that the execution of the file is stuck, and that the buffer is full, maybe the buffer can't empty itself ?
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/8640162/logfile.txt)

Have you ever seen this happen before ? Your hindsight would be useful for me to understand if the problem comes from our end (potentially something related to electromagnetic compatibility or a problem with our card) or not.

Thank you for your time

svenhb, 2022-05-07:
No idea, without extended logging....
What is see is:
strange received data on 3rd serial: 
```
2022-05-05 15:56:27.7610 | Trace | GrblPlotter.SimpleSerialForm   | 3rd serial RX '??m??@XZm???<???:??S[?>?=[? ????X-???&???t6Vx%h9wY???u>??]??Qr??3Q??oa??=#?1Z??%V{-?0?:@T?)|3?Xp}???Ni?\k?}??@N?v_|??v??s??8??f1??\????K9OH^z?`???Y?ZU?=rX?%???<????y?E????FR?'????@O??'   busy:True
2022-05-05 15:56:27.7610 | Trace | GrblPlotter.SimpleSerialForm   | 3rd serial RX '!??o*???O?aq?)???"6?0?*??nT7?h?y?s?T?u"]0?'   busy:True
```
3rd serial toggles busy-signal, but 1st serial doesn't start further sending:
```
2022-05-05 16:06:20.1140 | Trace | GrblPlotter.SimpleSerialForm   | 3rd serial RX 'soit : 999.99 mm'   busy:True
2022-05-05 16:06:20.1140 | Trace | GrblPlotter.SimpleSerialForm   | 3rd serial RX 'READY'   busy:False

```

YBougeard, 2022-05-16:
Thank you for your feedback, I'll look into that.

svenhb, 2022-05-28:
Please check new version 1.6.6.5. I added more logging:
If there is no status change during streaming, I log some variables, to find the problem (StreamingMonitor)

svenhb, 2022-08-11:
Seems to work now

closed 2022-08-11 by svenhb