svenhb/GRBL-Plotter #288: Problem with tool change on coreXY pen plotter

anbello, 2022-05-25:
**Describe the bug**
I have a coreXY pen plotter with grbl-coreXY 0.9i on Arduino nano and it is near all working OK with GRBL-Plotter.
The problem is on multi group plotting and tool change. I have set to change the tool manual (Pause M0) but when a new group begin the pen goes up then appear the dialog box 'Tool change needed' and then immediately pen goes down. This last pen down should not happen and is a problem.

For pen up / down I have a servo controlled by 'M03 S50' for up and 'M03 S0' for down.

svenhb, 2022-05-25:
Please check this setting:
I assume the PWM will be switched off via M5 with the result, that the pen goes down...
[image]

svenhb, 2022-05-25:
Yes, sure it is ok to uncheck. It is for router with spindel, if you need to change the bit.

closed 2022-05-25 by anbello