svenhb/GRBL-Plotter #289: Problem with feed slow down on large gcode file

anbello, 2022-05-25:
**Describe the bug**
The problem is with my coreXY pen plotter with grbl-coreXY 0.9i on Arduino nano.
With a  gcode of near 100000 lines (produced by many tiny circle) when the plotting is near half complete all the gcode lines has been read by the program and the plotter continue to plot (I think using some sort of buffer) but at a very low speed.

svenhb, 2022-05-25:
Can you show the output of the COM-window?
What do you mean with "...and the plotter continue to plot", did it stop before because of tool-change?

anbello, 2022-05-25:
It is a single color plot, no tool change. On the left panel I see all the lines has been read but the plot is near the half, it continues to plot but at low speed.
Now I can't post the output, I will do asap.

svenhb, 2022-05-25:
Perhaps you can post the gcode from the left panel?

svenhb, 2022-05-26:
With my Arduino nano on the desk, with grbl 1.1 I could also notice the slow down...
After pressing pause, the speed came back...
No idea for the moment...

I would use different import settings:
For circles I would use G2/G3 commands 
and the N-line numbers are not needed....

Uncheck:
[image]
[image]

anbello, 2022-05-26:
Thanks, I will try and will let you know.

svenhb, 2022-05-26:
I don't have your original SVG or DXF to import it with other settings...

anbello, 2022-05-26:
Sorry, here it is.
[boids2d-red.svg.zip](https://github.com/svenhb/GRBL-Plotter/files/8780721/boids2d-red.svg.zip)

svenhb, 2022-05-26:
Ok, too bad, the SVG contains paths in circle shapes, but not real circles, no way to reduce code-size here.
But you could avoid tiny moves, which reduces line-count from 95424 to 38408:
[image]
In 2D view it looks now a bit angular, but I'm sure the real output looks still good:
[image]

anbello, 2022-05-27:
Thanks a lot, I didn't yet tried because the pen plotter is not here with me, I will try today and let you know.

anbello, 2022-05-27:
Sorry, another question:
Is it normal this behavior with long gcode files or is it something that will be addressed in the future?

svenhb, 2022-05-27:
I'm working on it

svenhb, 2022-05-28:
Please check new version: https://github.com/svenhb/GRBL-Plotter/releases

anbello, 2022-05-29:
I have tested new version, with 'Remove very short moves' < 0.1 it has random slow down and return to normal speed during the plotting, with < 0.5 all OK from start to finish and circles don't seems segmented to the eyes.

Thanks

svenhb, 2022-05-29:
Thanks for the feedback. So the main problem: "after finish sending the code from left panel the plotter slows down" is solved.

>with 'Remove very short moves' < 0.1 it has random slow down and return to normal speed 

Probably too much commands per seconds needed, to keep the grbl-firmware buffer filled...
At feedrate 1000 mm/min ca. 167 command-lines with ca. 15 chars each needs to be send and visulized in GRBL-Plotter.

closed 2022-05-29 by svenhb

svenhb, 2022-05-29:
>Probably too much commands per seconds needed

You could reduce command length by use of less decimals (just 1 instead of 3):
[image]

By use of "compress" and / or "Relative movements":
[image]