svenhb/GRBL-Plotter #290: Automatic scaling doesnt work on big image

yagizzha, 2022-05-27:
[image]

Circle is detected and there is no "fiducial not detected" error , also tried centering the circle with shape recognition and that went correctly too however second fiducial (the one shown in image) is not getting correctly recognized . First one is working as intended.

svenhb, 2022-05-27:
You need to adjust the filters for your needs:
1. enable filter result
2. adjust filter until your fiducial is recognized (red or yellow)
3. save new settings
[image]

svenhb, 2022-05-28:
Any improovements?

yagizzha, 2022-06-08:
Couldnt try yet , will let you know as soon as we get a result

svenhb, 2022-08-11:
Any improvements?

svenhb, 2023-01-05:
I hope it is fixed

closed 2023-01-05 by svenhb