svenhb/GRBL-Plotter #291: servo motor command

ali-allafi-1996, 2022-05-28:
hello, I have some questions for you 
1- how I can adjust the program ( GRBL Plotter) to don't generate the command (M03/M04),
 when it generates the code this command is automatically generated with the g code.
I don't use the servo motor except for the Gripper when the pen change.
please I want to control my g code generated, I want to code without the (M03/M05) command.
2- the homing cycle in this program how I can fix my homing position for x, y, and z coordinate in this program?

svenhb, 2022-05-28:
You need to adjust your pen-up/-down commands here: If you don't want M03/M04, don't choose Laser or PWM, but Z axis or put-in your own commands (individual). With Z-axis, you may enable "Don't use spindle", to avoid inital M03.
[image]

Homing cycle and position is inside grbl-firmware, no chance to change.
After homing you get not changeable machine-coordinates - but who cares. You will use work-coordinates (except for tool-changes): https://grbl-plotter.de/index.php?id=auto-pen-change

ali-allafi-1996, 2022-05-28:
[image]

ali-allafi-1996, 2022-05-28:
I don't use the servo as z axes but I used it in the tool changer as a gripper.
I don't need this command in my main g code.
how I can avoid it?

ali-allafi-1996, 2022-05-28:
[image]

svenhb, 2022-05-28:
Are you using latest version? As you can see, no M3/M4 command:
[image]

svenhb, 2022-05-28:
Using individual commands:
[image]

ali-allafi-1996, 2022-05-28:
could you give me the setting to use it on my own 
[image]

ali-allafi-1996, 2022-05-28:
I have another problem with the tool changer.
[image]
 how I can use it. because I want to add a special code to the black and blue colors.

svenhb, 2022-05-28:
Just click into the cell and put in your code

ali-allafi-1996, 2022-05-28:
[image]
is it the right way to use it?

ali-allafi-1996, 2022-05-28:
m03 for the black color 
m05 for the blue color 
that is what I want.

svenhb, 2022-05-28:
After changing, you need to save and reload tool table example "myTools"
[image]

You need to enable grouping by colors and activate tool table:
[image]

Then code will be applied:
[image]

ali-allafi-1996, 2022-05-28:
Okay, 
I think i don't use this command after now
[image]
because I use a special command from the table is it right?

svenhb, 2022-05-28:
If you don't need scripts for tool change (see below), you need not to enable.
[image]

ali-allafi-1996, 2022-05-28:
Okey my friend thanks for giving me all the answers that I want . 
Now I can turn on my CNC plotter as I want it .

closed 2022-05-29 by ali-allafi-1996