svenhb/GRBL-Plotter #292: GRBL.exe V1655 Virus Scan

mortonhoby, 2022-05-31:
On downloading the latest version (v1665), my Antivirus software is flagging several files (mainly the GRBL-Plotter.exe and several of the GRBL-Plotter.resources.dll files in each of the language folders) that are marked as having the threat named 'WS.reputation.1'.   Not sure if this is a false +ve but would like to know if anyone else has had this flagged and if indeed they have experienced an problems if they exclude it from future scans.

svenhb, 2022-06-01:
Bad to hear, I found this (translated from german to english):

_WS.Reputation 1 during virus check: What is that?
In most cases, you can stay calm if a file is immediately tagged with the "WS.Reputation 1" message. As a rule, it is not a virus, but a “false positive”.

Rather, this feature is given to files that are very uncommon among users.
This gives an unknown reputation for the functionalities of this file.
Antivirus programs examine the behavior of individual files in order to include services with malicious behavior in the virus database.
For files with "WS.Reputation 1" there are only too few user signals to be able to make a reliable statement about the security.
A virus can, but does not have to be present._

If you are not happy with this statement, you could create the exe by yourself using the source code...

mortonhoby, 2022-06-01:
Thank you Sven.

 

I have only seen this message once before and indeed that was a piece of software that didn’t have many user (according to Norton), so your comments seem to fit.

 

I feel comfortable now if removing the files from Quarantine and runing the latest version.

 

At some point I will take a look at the source code but for now your comments are much appreciated.

 

Many Thanks

 

Roger Morton

closed 2022-08-11 by svenhb