svenhb/GRBL-Plotter #294: Hi how can I put space before Y line (or end of X line)

corumlee, 2022-06-16:
If I convert IMG in to gcode and create G-Code.

Then happens like attached pic

So my mpcnc machine doesn't recognize the command.

Can somebody help this?

[image]

svenhb, 2022-06-16:
You could use the "replace" command from the editor:
Right click:
[image]

Then replace "Y" by " Y"
[image]

corumlee, 2022-06-16:
Thank you very much 

It is working now

closed 2022-08-11 by svenhb