svenhb/GRBL-Plotter #296: tangential knife and pen.

jbaker1311, 2022-07-05:
I'm new to the software and was wondering how to create or modify a use case/tools 
Specifically I'm planing to use a rotary cutting wheel on z axis, with a pen offset 150mm from said wheel.

Also, does this the software support the use of air assist lifting of pen/cutter?

svenhb, 2022-07-05:
Just do your settings as needed...
Sounds like, you want to use a regular Arduino with XYZ axis, so I would start with this setup (Example 2): https://grbl-plotter.de/index.php?id=tangential-knife&setlang=en
And here you can set your individual command for lifting the pen: https://grbl-plotter.de/index.php?id=form-setup#Gcodegeneration

So you may use the pin for spindle (M3,4,5) or coolant (M7,8,9) to switch your air assist: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

jbaker1311, 2022-07-06:
Thank you for the swift reply, I'll be testing it this weekend so will more than likely be asking more questions.

On a side note, when loading in a dxf file with specific line colours for marking/cutting, I think the software isn't receiving the colours for seperate tools

svenhb, 2022-07-06:
Here is a description using a SVG, but should also work with DXF: https://grbl-plotter.de/index.php?id=laser-pen&setlang=en
GRBL-Plotter can seperate by layer, color or pen-width.

svenhb, 2022-07-07:
Perhaps you can show your setup?

svenhb, 2022-07-07:
Ok, I think the problem is: You would need two different pen-up/down signals for your two tools. 

My idea: Control via "coolant" (M7,8,9)  which tool is selected and control pen-up/down via spindle on/off.
Inside the tool table you apply M7 of M9 depending on needed tool (and also G54/G55 if there is an offset to apply).

jbaker1311, 2022-07-07:
I'm still not sure on how to program the offset of the tools, as it the tools aren't static.
Basically, the rotary cutter in 0,0,0 the pen is at 0,100,0
When the rotary cutter moves to 150,150,0 the pen would be at 150,250,0

The pen up/down is still in lieu as I'm waiting for air fittings.
I'll keep playing with it and test tomorrow.

svenhb, 2022-07-07:
>When the rotary cutter moves to 150,150,0 the pen would be at 150,250,0
 
This you can compensate with the different coordinate systems, e.g. G54, G55.

jbaker1311, 2022-07-26:
G'day Sven, I've worked out how to offset the rotary cutter and employ the relays to use air assist, now I'm having trouble distinguishing between lowering the pen and cutter

svenhb, 2022-07-26:
As written above:
My idea: Control via "coolant" (M7,8,9) which tool is selected and control pen-up/down via spindle on/off.
Inside the tool table you apply M7 of M9 depending on needed tool (and also G54/G55 if there is an offset to apply).

jbaker1311, 2022-07-28:
Sorry, do you have an example on how to write them into different tools? Do I use the pen up pen down translation dialogue or the tool table?

svenhb, 2022-07-28:
I thought like this:
commands seperated by semicolon ';' to select the tool. To use a tool, setup the pen-up-down-translation dialogue.
[image]

jbaker1311, 2022-07-28:
Okay I'll have a go with it like that, I've currently got it working about 80% about to have the vacuum table connected so it'll be having a workout soon

jbaker1311, 2022-07-28:
No success, with nothing in the individual commands and tool table with commands they both lower when only one is required, I also want to turn off tangential z axis with pen and have it enabled when needed to cut

svenhb, 2022-07-29:
Do it work is you send the commands manually?

- M7 to select tool 'a', then M3 to lower, M5 to lift
- M9 to select tool 'b', then M3 to lower, M5 to lift

Perhaps you have some pictures and wiring diagram?
What is your hardware? Arduino nano with CNC-Shield?

jbaker1311, 2022-07-29:
So when selected above the gcode editor, it works.
I don't have any pictures though I'm using a Mega with a pinout board. 2 channel relay for activating the solenoids on the gantry.

svenhb, 2022-07-29:
>So when selected above the gcode editor, it works.

Sounds good.
How does the created G-Code look like? Can you find the corresponding commands?

jbaker1311, 2022-07-29:
I'm not at the workshop to give you exactly what it looks like but it shows g55 m8 then when going through the streaming it says m8 (pu)
I'll try recreate it on my home computer and get back to you with actual gcode readout

svenhb, 2022-07-29:
With this as a pattern: https://grbl-plotter.de/index.php?id=laser-pen
I assume you do tool up/down only with M3/M5, no Z-axis - correct?
Then use this modifications:
[image]
Don't forget to save modified tool-table:
[image]

Result should look like:
[image]
and
[image]

svenhb, 2022-07-29:
>I'm using a Mega

If you use grbl version "grbl-Mega-5X" you could also use digital outputs, controllable via M62, 63, 64, 65.
https://github.com/fra589/grbl-Mega-5X

svenhb, 2022-08-11:
Seems to work

closed 2022-08-11 by svenhb