svenhb/GRBL-Plotter #297: 3d Printing svg files with filament per vector

kneo1, 2022-07-10:
Hi. I am experimenting with 3d printing  an svg file. One line per vector and only 1 "slice" or level.
Simple lines to be printed directly by the 3d printer head.
I am using the "Command extension" with "Add distance related command 1" I use "B" letter save gcode and then search/replace B to E in notepad. 

About the "Factor" option, what should it be for extruding through a 0.4mm nozzle? With the default "1" it extrudes way too much.
I have tried 0.05 and it extrudes much more reasonably. 
1 mm extruded at the nozzle of 0.4mm equals 0.05224489795918367346938775510204 mm on E axis at 1.75mm
I believe the "Factor" only takes 2 decimal points.

For the filament retract I use "Pen up/down translation" with "Z" and "Individual" 
in "Z" heigh set/engrave to 2 and 0
in "Individual" "Pen Up/Down" G91;G1 E-4 F2000;G90 / G91;G1 E4 F2000;G90 for filament retraction on Z up movement. 
I switch to relative (G91) to retract since there is no way to know the absolute positioning on E axis.

With these options printing the SVG on 3d printer kind of works but extrude ratio is way off, too much filament.
About the "Command extention" "Add distance related command 1" the "Absolute value" checkbox does not seem to work? Makes little difference in gcode if any, idk.

Extruder calculations:
1.75mm filament area is 2.4052818754046854481979613403234 sq mm
0.4mm nozzle area is 0.12566370614359172953850573533118 sq mm
Ratio: 19.140625

So 1 (one) length of 1.75mm filament will be 19.140625 long 0.4mm filament
1 mm length of 0.4mm filament equals 1/19.140625 of the 1.75mm filament or 0.05224489795918367346938775510204 mm

The extrusion of the filament speed must match the XY speed of the extruder nozzle for the filament to remain relatively uniform 0.4mm If the XY movement is slower that the extrusion rate then more filament will be deposited and the filament will be squished between the nozzle and the table.

kneo1, 2022-07-12:
It looks like that (PD) is from the custom code one can write on the Pen up or down action. 
So I put G91 at the end and the relative mode stays on.

svenhb, 2022-07-12:
(PU/PD) are just internal markers, to be able to show the pen-up/-down path even if there is no pen-up/-down translation is given in the setup.

>So I put G91 at the end and the relative mode stays on.

Yes... but the X and Y values are still given in absolute coordinates - would give chaos...

Will try to fix that

closed 2023-05-01 by svenhb